ichi4096/vivado-on-silicon-mac #1: vivado crashes when i launch bitstream, below is the terminal log

shahbaazlokh, 2023-04-15:
Last login: Sun Apr 16 02:19:50 on ttys002
shaz@shahbaazs-MacBook-Air ~ % while ! [[ $(ps aux | grep vivado_container | wc -l | tr -d "\n\t ") == "1" ]]; do /Users/shaz/Downloads/vivado-on-silicon-mac-main/xvcd/bin/xvcd; sleep 1; done; exit
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed

Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.

[Process completed]

ichi4096, 2023-04-16:
I've encountered this behaviour before and it seems to have something to do with shared libraries and udev as seen [here](https://support.xilinx.com/s/question/0D54U00005Sgst2SAB/failed-batch-mode-execution-in-linux-docker-running-under-windows-host?language=en_US) and [here](https://community.flexera.com/t5/InstallAnywhere-Forum/Issues-when-running-Xilinx-tools-or-Other-vendor-tools-in-docker/m-p/245820#M10647). However, I've set the LD_PRELOAD variable as an environment variable in the Dockerfile, which fixed it on my machine. I've added into the start_vivado.sh script that it should load the settings64.sh script; try again and let me know if this helps (you can simply replace start_vivado.sh with the new version on Github, no need to reinstall Vivado again). If this issue persists, could you send me an example of a design that makes Vivado crash?

shahbaazlokh, 2023-04-16:
hi,

thanks for quick reply,

i tried to update start_vivado.sh script and its same problem.
i have attached design_1 tcl file, before you run this file make sure to rename this file as design_1.tcl and source this file after you launch vivado.

[design_1.txt](https://github.com/ichi4096/vivado-on-silicon-mac/files/11243098/design_1.txt)

ichi4096, 2023-04-16:
When I source that file i get

ERROR: [Board 49-71] The board_part definition was not found for xilinx.com:kr260_som_som240_2_connector_kr260_carrier_som240_2_connector_som240_1_connector_kr260_carrier_som240_1_connector:part0:1.1. The project's board_part property was not set, but the project's part property was set to xck26-sfvc784-2LV-c. Valid board_part values can be retrieved with the 'get_board_parts' Tcl command. Check if board.repoPaths parameter is set and the board_part is installed from the tcl app store.

in the Tcl console. Do I need to install something additional for this to work?

shahbaazlokh, 2023-04-16:
yes, there is a problem with this board selection, i have seen this on my linux machine as well,
no i have not installed anything manually.

however, i made a seprate example design with ac701 board with tcl script, attached below.

try this and let me know if the script can run and make block design.

[design_2.txt](https://github.com/ichi4096/vivado-on-silicon-mac/files/11244340/design_2.txt)

ichi4096, 2023-04-17:
I was able to open this design and it also crashed with that error message, which is weird since the environment variables (specifically LD_PRELOAD) from the Dockerfile should prevent that. It turns out that some of those variables are not passed to the non-root user, which starts Vivado.

I've added the appropriate lines into the start_vivado.sh script, so try it with the new one again. Also, increase the Swap in Docker dashboard -> Settings -> Resources to 2 GB, since Docker assigns too little by default and Vivado will crash otherwise.

Additionally, execute the command `defaults write org.xquartz.X11 enable_iglx -bool true` in your terminal once. This should prevent any graphics glitches, which I discovered during synthesizing.

shahbaazlokh, 2023-04-17:
hi,

thanks for this fix,
i was able to make bitstream.

tomorrow i am going to try hello world and led blink with arty spartan 7, that will use vivado and vitis for microblaze app with uart terminal.

thanks

closed 2023-04-17 by shahbaazlokh