ichi4096/vivado-on-silicon-mac #3: how to launch vitis_hls

shahbaazlokh, 2023-04-17:
i am adding start_vitis_hls script here, but it did ow worked on my system.
can you please add launch_vitis_hls.app same as launch_vivado

#!/bin/bash
cd /home/user\
export LD_PRELOAD="/lib/x86_64-linux-gnu/libudev.so.1 /lib/x86_64-linux-gnu/libselinux.so.1 /lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0"\
export JAVA_TOOL_OPTIONS="-Dsun.java2d.xrender=false"\
export JAVA_OPTS="-Dsun.java2d.xrender=false"\
export DISPLAY="host.docker.internal:0"\
/home/user/Xilinx/Vitis_HLS/*/settings64.sh\
/home/user/Xilinx/Vitis_HLS/*/bin/vitis_hls\
}

however i am not sure about LD_PRELOAD

thanks

ichi4096, 2023-04-17:
I'll add additional apps such as Launch_Vitis, but starting vitis is not possible from the terminal right now, so I'll have to fix that first.

shahbaazlokh, 2023-04-17:
i saw its not downloaded and installed, we will first have to download it and install it by adding in install_config.txt file

ichi4096, 2023-04-18:
Still, right now, vitis_hls does not run inside of this docker container. I'll have to fix it.

ichi4096, 2023-05-02:
The underlying error is the same as in issue #4.

closed 2023-05-02 by ichi4096