ichi4096/vivado-on-silicon-mac #4: Simulation issue

a-ricci, 2023-04-18:
Are you able to run any simulations? It mentions permissions, but that is not the issue. In compile.log I see the following:

```
assertion failed [metadata != nullptr]: request to change permissions for interval that isn't tracked
(VMAllocationTracker.cpp:469 change_permissions)
/tools/Xilinx/Vivado/2022.2/bin/rdiArgs.sh: line 312: 3424 Trace/breakpoint trap "$RDI_PROG" "$@"
```

arminsch, 2023-04-18:
BTW, @ichi4096: Thanks a lot for your altruism to share your efforts.

ichi4096, 2023-04-18:
Thanks @arminsch :-) 

I've only tested Vivado by editing, sythesizing, running implementation, generating bitstream and programming devices because that's all we had to do in our classes for now.

But I'll try to debug this issue, which is also seen when starting vitis_hls; my guess is that it doesn't work because of some Docker or Rosetta related issues, since Vivado works in virtual machines otherwise...

The cpp file says somthing with Allocation and I remember somthing about different malloc implementations when going through the Vivado shell scripts, so I have a starting point of investigation.

arminsch, 2023-04-18:
Maybe also giving some hints:
* `xar`, `xcrg`, `xelab`, `xsc`, `xvhdl`, `xvlog` all raise the same exception mentioned above)
* `xtclsh`and `xsdb` raise a somewhat relate exception
~~~
assertion failed [result.value != EEXIST]: VmTracker attempted to allocate existing mapping
(ThreadContextVm.cpp:47 mmap)
/opt/Xilinx/Vivado/2022.2/bin/xtclsh: line 72:  2543 Trace/breakpoint trap   (core dumped) "$RDI_BINROOT"/unwrapped/"$PLATFORM_TMP$RDI_OPT_EXT"/rlwrap -rc "$RDI_BINROOT"/loader -exec tclsh8.5 "${RDI_ARGS[@]}"
~~~

arminsch, 2023-04-18:
Let me know in case you need someone to discuss. I'm eager to help as this seems to be the last remaining issue that prevents me from changing my Laptop-HW.

ichi4096, 2023-04-18:
Well I tried strace and ltrace on the vivado_hls binary (in the unwrapped folder) and both seem to be broken under Rosetta:

```
user@3fa853c311c2:~/Xilinx/Vitis_HLS/2022.2/bin$ export LD_LIBRARY_PATH=/home/user/Xilinx/Vitis_HLS/2022.2/lib/lnx64.o/Ubuntu:/home/user/Xilinx/Vitis_HLS/2022.2/lib/lnx64.o:/home/user/Xilinx/Vitis_HLS/2022.2/tps/lnx64/jre11.0.11_9/lib/:/home/user/Xilinx/Vitis_HLS/2022.2/tps/lnx64/jre11.0.11_9/lib//server:/home/user/Xilinx/Vitis_HLS/2022.2/bin/../lnx64/tools/dot/lib
user@3fa853c311c2:~/Xilinx/Vitis_HLS/2022.2/bin$ strace ./unwrapped/lnx64.o/vitis_hls 
[ Process PID=511 runs in x32 mode. ]
[ Process PID=511 runs in 64 bit mode. ]
[ Process PID=511 runs in x32 mode. ]
[ Process PID=511 runs in 64 bit mode. ]
[ Process PID=511 runs in x32 mode. ]
[ Process PID=511 runs in 64 bit mode. ]
[ Process PID=511 runs in x32 mode. ]
[ Process PID=511 runs in 64 bit mode. ]
[ Process PID=511 runs in x32 mode. ]
[ Process PID=511 runs in 64 bit mode. ]
[ Process PID=511 runs in x32 mode. ]
[ Process PID=511 runs in 64 bit mode. ]
[ Process PID=511 runs in x32 mode. ]
[ Process PID=511 runs in 64 bit mode. ]
[ Process PID=511 runs in x32 mode. ]
[ Process PID=511 runs in 64 bit mode. ]
[ Process PID=511 runs in x32 mode. ]
[ Process PID=511 runs in 64 bit mode. ]
[ Process PID=511 runs in x32 mode. ]
[ Process PID=511 runs in 64 bit mode. ]
[ Process PID=511 runs in x32 mode. ]
[ Process PID=511 runs in 64 bit mode. ]
[ Process PID=511 runs in x32 mode. ]
[ Process PID=511 runs in 64 bit mode. ]
[ Process PID=511 runs in x32 mode. ]
[ Process PID=511 runs in 64 bit mode. ]
[ Process PID=511 runs in x32 mode. ]
[ Process PID=511 runs in 64 bit mode. ]
[ Process PID=511 runs in x32 mode. ]
[ Process PID=511 runs in 64 bit mode. ]
[ Process PID=511 runs in x32 mode. ]
[ Process PID=511 runs in 64 bit mode. ]
[ Process PID=511 runs in x32 mode. ]
[ Process PID=511 runs in 64 bit mode. ]
[ Process PID=511 runs in x32 mode. ]
[ Process PID=511 runs in 64 bit mode. ]
[ Process PID=511 runs in x32 mode. ]
[ Process PID=511 runs in 64 bit mode. ]
[ Process PID=511 runs in x32 mode. ]
[ Process PID=511 runs in 64 bit mode. ]
[ Process PID=511 runs in x32 mode. ]
[ Process PID=511 runs in 64 bit mode. ]
[ Process PID=511 runs in x32 mode. ]
[ Process PID=511 runs in 64 bit mode. ]
[ Process PID=511 runs in x32 mode. ]
[ Process PID=511 runs in 64 bit mode. ]
[ Process PID=511 runs in x32 mode. ]
[ Process PID=511 runs in 64 bit mode. ]
[ Process PID=511 runs in x32 mode. ]
[ Process PID=511 runs in 64 bit mode. ]
[ Process PID=511 runs in x32 mode. ]
[ Process PID=511 runs in 64 bit mode. ]
[ Process PID=511 runs in x32 mode. ]
[ Process PID=511 runs in 64 bit mode. ]
[ Process PID=511 runs in x32 mode. ]
[ Process PID=511 runs in 64 bit mode. ]
[ Process PID=511 runs in x32 mode. ]
[ Process PID=511 runs in 64 bit mode. ]
[ Process PID=511 runs in x32 mode. ]
[ Process PID=511 runs in 64 bit mode. ]
[ Process PID=511 runs in x32 mode. ]
[ Process PID=511 runs in 64 bit mode. ]
[ Process PID=511 runs in x32 mode. ]
[ Process PID=511 runs in 64 bit mode. ]
[ Process PID=511 runs in x32 mode. ]
[... 89 lines cut ...]
[ Process PID=511 runs in x32 mode. ]
[ Process PID=511 runs in 64 bit mode. ]
[ Process PID=511 runs in x32 mode. ]
[ Process PID=511 runs in 64 bit mode. ]
[ Process PID=511 runs in x32 mode. ]
[ Process PID=511 runs in 64 bit mode. ]
[ Process PID=511 runs in x32 mode. ]
[ Process PID=511 runs in 64 bit mode. ]
[ Process PID=511 runs in x32 mode. ]
[ Process PID=511 runs in 64 bit mode. ]
[ Process PID=511 runs in x32 mode. ]
[ Process PID=511 runs in 64 bit mode. ]
[ Process PID=511 runs in x32 mode. ]
[ Process PID=511 runs in 64 bit mode. ]
assertion failed [metadata != nullptr]: request to change permissions for interval that isn't tracked
(VMAllocationTracker.cpp:469 change_permissions)
+++ killed by SIGTRAP +++
Trace/breakpoint trap
user@3fa853c311c2:~/Xilinx/Vitis_HLS/2022.2/bin$ ltrace ./unwrapped/lnx64.o/vitis_hls 
"/proc/513/exe" is ELF from incompatible architecture
user@3fa853c311c2:~/Xilinx/Vitis_HLS/2022.2/bin$ assertion failed [metadata != nullptr]: request to change permissions for interval that isn't tracked
(VMAllocationTracker.cpp:469 change_permissions)
sudo: handle_sigchld_pty: waitpid: No child processes
```
Maybe debugging with gdb will help...

a-ricci, 2023-04-19:
@ichi4096, thanks for the quick feedback! I'm looking it this as well

ichi4096, 2023-04-19:
I figured out how to run `xtclsh` and `xsdb`: Both scripts open the underlying binary (in `unwrapped` folder) with a software called `rlwrap`. Installing it separately with apt and replacing the binary in the unwrapped folder with a symlink to the downloaded version of `rlwrap` makes these two programs work.

a-ricci, 2023-04-20:
I see. I tried to debug this a bit but no luck on my side as well. I does seem like a Rosetta-related issue. Quite a few people are getting problems when doing mamps: https://github.com/docker/roadmap/issues/384

ichi4096, 2023-04-24:
Sorry for the delay, I had some studying to do. The good news: I finally figured out how to debug programs using the RosettaLinux `rosetta` binary with which x64 ELF binaries are run (using UTM, not Docker).

When running it without arguments, it shows this:

```
debian@debian:/home/user/Xilinx/Vivado/2022.2/bin/unwrapped/lnx64.o$ /media/rosetta/rosetta 
Usage: rosetta <x86_64 ELF to run>

Optional environment variables:
ROSETTA_DEBUGSERVER_PORT    wait for a debugger connection on given port

version: Rosetta-289.7
```

Googling the name of the environment variable led to [this blog](https://sporks.space/2023/04/12/debugging-an-x86-application-in-rosetta-for-linux/), which explained the whole process.

And instead of getting [meaningless translated garbage](https://pastebin.com/LVaACBxv), with which I can do pretty much nothing since it's JIT-compiled, I get actual x64 assembly :O As was to be expected, there are no debug symbols, but I'll see how far I get.

ichi4096, 2023-04-27:
There is very little I can do without debug symbols... So I asked for them [here](https://support.xilinx.com/s/question/0D54U00006nUESzSAO/debug-symbols-for-libisliostreamsso?language=en_US). Probably, I won't get an answer, however, I can ask specifically for the source code of the `rlwrap` binary since it's GPL licensed. If I can reproduce the error with the source code, maybe I'll get closer to the issue...

ichi4096, 2023-07-27:
So… I got a response on the Xilinx support forum, but it doesn‘t help. This issue seems to be specifically rosetta-related, so I reported this issue via the Feedback app in macOS.

In the meantime, somewhere in the EULA, they mention the version of rlwrap they used, which I downloaded via the wayback machine. I‘ll try to compile it with debug symbols and debug it properly.

Also sorry for not updating for a long time, I had some catching up to do with my studies.

ichi4096, 2023-08-20:
The simulation should work on the 2023 version, for which I just added support. Can anyone confirm? At least the binaries mentioned above don't crash anymore.

WindHurtLZ, 2023-09-03:
> The simulation should work on the 2023 version, for which I just added support. Can anyone confirm? At least the binaries mentioned above don't crash anymore.

---
I got same error when i simulate on 2023 version.

assertion failed [metadata != nullptr]: request to change permissions for interval that isn't tracked
(VMAllocationTracker.cpp:469 change_permissions)
/home/user/Xilinx/Vivado/2023.1/bin/rdiArgs.sh: line 352:   282 Trace/breakpoint trap   "$RDI_PROG" "$@"

henriquenunez, 2023-09-12:
Hey ppl, I also came across this problem now when trying to launch a simulation (pretty simple now, was just to see how to use vivado itself).

For me the problem is the following message:
[image]

When running the simulation from the following button:
[image]

I am here to help fixing the issue, if needed, just could use some guidance XD.

ichi4096, 2023-09-23:
Could anybody please give a concrete example of a simulation that doesn't run? Meaning include the files, Vivado version, macOS version, Docker version and error messages as well as steps to reproduce the error.

CasparB, 2023-10-03:
Hey, I'm having this issue on Vivado v2023.1, macOS Ventura 13.0.1, Docker v24.0.6. Error:

[image]

Console: 

[image]

Constraint file, source files:

[files.zip](https://github.com/ichi4096/vivado-on-silicon-mac/files/12797379/files.zip)

Please let me know if you need more info!

keegandent, 2024-01-08:
I can run post-synthesis functional sim but not post-synthesis timing sim, basically the same errors as CasparB.

Really appreciate your work on this @ichi4096. Let me know if you need another test project for comparison on this.

ichi4096, 2024-07-29:
Could someone try this with the recent version of this tool and macOS 15? This Rosetta issue may have been resolved.

ichi4096, 2024-08-02:
> Maybe also giving some hints:
>
>     * `xar`, `xcrg`, `xelab`, `xsc`, `xvhdl`, `xvlog` all raise the same exception mentioned above)
>
>     * `xtclsh`and `xsdb` raise a somewhat relate exception
>
>
> ``` [...]

These binaries seem to be able to run under macOS 15. If you encounter simulation issues with the new version of my scripts, please open a new issue.

closed 2024-08-02 by ichi4096