ichi4096/vivado-on-silicon-mac #5: Not enough space error

fred-corp, 2023-04-18:
I'm having the following error message when I try to install the container :

```zsh
You successfully logged into your account. The installation will begin now.
If a window pops up, simply close it to finish the installation.
This is a fresh install.
Running in batch mode...
Picked up JAVA_TOOL_OPTIONS: -Dsun.java2d.xrender=false
Copyright (c) 1986-2023 Xilinx, Inc.  All rights reserved.
INFO  - User has accepted the EULAs. 
ERROR - There is not enough disk space on /home/user/Xilinx to install. 
```

Despite having ~85GB free on my hard drive. Do I need even more ? README says it should only download ~20GB..

ichi4096, 2023-04-18:
The following command (run in terminal) will show how much space is avaliable per mount:

`/usr/local/bin/docker run --rm --name vivado_container --mount type=bind,source="/tmp/.X11-unix",target="/tmp/.X11-unix" --mount type=bind,source="/Users/$USER/Downloads/vivado-on-silicon-mac-main",target="/home/user" --platform linux/amd64 -it x64-linux sudo -H -u user df -h`

What does it output for you?

fred-corp, 2023-04-18:
This is the output :
```zsh
Filesystem      Size  Used Avail Use% Mounted on
overlay         125G   11G  108G  10% /
tmpfs            64M     0   64M   0% /dev
shm              64M     0   64M   0% /dev/shm
grpcfuse        466G  385G   82G  83% /tmp/.X11-unix
/dev/vda1       125G   11G  108G  10% /etc/hosts
tmpfs           2.0G     0  2.0G   0% /proc/acpi
tmpfs           2.0G     0  2.0G   0% /sys/firmware
```

I've already checked in the docker settings panel to allocate as much virtual disk as possible

fred-corp, 2023-04-18:
Okay I've moved some files to an external drive to have around 95GB of free space and it seems to work now..
The offline installer on Xilinx website seems to take 89.4GB, so I guess that's how much you need :| )

closed 2023-04-18 by ichi4096

fred-corp, 2023-04-18:
Indeed ! By the way I'm still using an Intel MacBook Pro, so if this works the container will be validated on Intel as well :p 
How large is the container once everything is set up ?

ichi4096, 2023-04-18:
The Docker image is about 1 GB, the Vivado installation takes around 55 GB on my machine. Also, I noticed your `df -h` output generated some lines not present on my machine, so I was already suspicious about you using an Intel Mac ;-).

One thing that definetly will not work on your MacBook is xvcd (see readme) because it is compiled for Arm Macs. It could be recompiled for Intel Macs, but I don't have access to one to do that. This means that for the time being, you won't be able to program any devices via USB.

If this is somthing you need, consider opening another issue for that.