ichi4096/vivado-on-silicon-mac #6: Launching without using launch_vivado.app

adanowitz, 2023-04-28:
My organization locks down my mac so it can't run any .app's unless they're from apple or a trusted developer, and I don't have permission to override that in privacy and security. Is there a way to do what the app does from command line?

adanowitz, 2023-04-28:
Nevermind. Reading the install scripts found that in the Download directory Launch_Vivado.app is a directory that contains a script "Launch_Vivado" that does what I need. Now I just need to find a way to get xvcd to run...

ichi4096, 2023-04-28:
Are privileges also required for running xvcd on it's own?

ichi4096, 2023-05-02:
Unless you're willing to pay for a $100/year Apple Developer license, this issue will remain unresolved because I have to sign the application in order for it to run without admin privileges.

closed 2023-05-02 by ichi4096