ichi4096/vivado-on-silicon-mac #7: Do not open launch_vivado.app agin after restart PC

Yao-Chin, 2023-05-06:
Follow your installation step, I could successfully enable launch_vivado.app.
However, I restart my mac computer again.
Launch this application can't implement. 

And show this message on terminal as below:
docker: Error response from daemon: invalid mount config for type "bind": bind source path does not exist: /host_mnt/private/tmp/.X11-unix.
See 'docker run --help'.
tab 1 of window id 241

Do you have any suggestion to solve it?
Thank your help

ichi4096, 2023-05-06:
This happens when Docker is opened before Launch_Vivado is run, but XQuartz hasn't been opened before. I'll have to add a fix for that in the launch script. In the meantime, you can simply start XQuartz manually before starting Launch_Vivado.

Yao-Chin, 2023-05-06:
I see my launch script as below.
----
#!/bin/zsh
open -a XQuartz
open -a Docker
while ! /usr/local/bin/docker ps &> /dev/null
do
open -a Docker
sleep 5
done

/usr/local/bin/docker run --rm --name vivado_container --mount type=bind,source="/tmp/.X11-unix",target="/tmp/.X11-unix" --mount type=bind,source="/Users/yao-chin/Documents/vivado-on-silicon-mac-main",target="/home/user" --platform linux/amd64 x64-linux sudo -H -u user bash /home/user/start_vivado.sh &
osascript -e 'tell app "Terminal" to do script " while ! [[ $(ps aux | grep vivado_container | wc -l | tr -d \"\\n\\t \") == \"1\" ]]; do /Users/yao-chin/Documents/vivado-on-silicon-mac-main/xvcd/bin/xvcd; sleep 1; done; exit"' 
----
The XQuartz had opened before the Docker.

Yao-Chin, 2023-05-06:
I try to remove XQuartz and install it again.
And then, Lanuch_vivado app will successfully operate.

ichi4096, 2023-05-06:
The open command is equivalent to launching an App with a mouse click. In the script, the open command terminates almost immediately, even when the App hasn't fully started yet. I have a while loop that checks wether Docker has started up, but there is no check for wether XQuartz is fully started up. In practice, Docker will take a longer time to start up, so when it isn't already started, XQuartz has enough time to start up in parallel with Docker.

In summary: Start XQuartz manually before running Launch_Vivado. I'll work on a fix in the meantime so that this isn't required.

ichi4096, 2023-05-06:
I have fixed the script now. However, you have to reinstall everything in order for the changes to be noticeable. Again, you could just start XQuartz before running Launch_Vivado, which will have the same effect.

closed 2023-05-06 by ichi4096