ichi4096/vivado-on-silicon-mac #8: Command line force stop after drop in the files

potatochick2020, 2023-05-18:
Bellow is everything I did
```
Launching Docker container and installation script
Installer was previously extracted. Removing the extracted directory.
Found Installer
Extracting installer
Creating directory /home/user/installer
Verifying archive integrity... All good.
Uncompressing AMD Installer for FPGAs and Adaptive SoCs..................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................
Log into your Xilinx account to download the necessary files.
This is a fresh install.
INFO Could not detect the display scale (hDPI).
       If you are using a high resolution monitor, you can set the insaller scale factor like this:
       export XINSTALLER_SCALE=2
       setenv XINSTALLER_SCALE 2
Running in batch mode...
Picked up JAVA_TOOL_OPTIONS: -Dsun.java2d.xrender=false
This command is not for general use and should only be run as the result of a call to
ProcessBuilder.start() or Runtime.exec() in a java application
Copyright (c) 1986-2022 Xilinx, Inc.  All rights reserved.
Copyright (c) 2022-2023 Advanced Micro Devices, Inc.  All rights reserved.
This command is not for general use and should only be run as the result of a call to
ProcessBuilder.start() or Runtime.exec() in a java application
ERROR - There was an error executing: [/bin/sh, -c, echo $TMPDIR]
INFO  - Internet connection validated, can connect to internet.
INFO  - In order to generate the authentication token please provide your AMD account E-mail Address and password.

E-mail Address:xxoo@gmail.com
Password:
INFO  - Generating authentication token...
INFO  - Saved authentication token file successfully, valid until 05/25/2023 02:21 PM

You successfully logged into your account. The installation will begin now.
If a window pops up, simply close it to finish the installation.
This is a fresh install.
INFO Could not detect the display scale (hDPI).
       If you are using a high resolution monitor, you can set the insaller scale factor like this:
       export XINSTALLER_SCALE=2
       setenv XINSTALLER_SCALE 2
Running in batch mode...
Picked up JAVA_TOOL_OPTIONS: -Dsun.java2d.xrender=false
This command is not for general use and should only be run as the result of a call to
ProcessBuilder.start() or Runtime.exec() in a java application
Copyright (c) 1986-2022 Xilinx, Inc.  All rights reserved.
Copyright (c) 2022-2023 Advanced Micro Devices, Inc.  All rights reserved.
INFO  - User has accepted the EULAs.
ERROR - The value specified in the configuration file for Modules (Virtex UltraScale+ HBM:1,DocNav:1,Kintex UltraScale:1,Artix UltraScale+:1,Spartan-7:1,Artix-7:1,Virtex UltraScale+:1,Vitis Model Composer(Xilinx Toolbox for MATLAB and Simulink. Includes the functionality of System Generator for DSP):1,Zynq UltraScale+ MPSoC:1,Zynq-7000:1,Virtex UltraScale+ 58G:1,Kintex-7:1,Install Devices for Kria SOMs and Starter Kits:1,Kintex UltraScale+:1) is not valid.

Generating App icon
./install.sh:68: no matches found: Xilinx/Vivado/*/doc/images/vivado_logo.png
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
icon.iconset:Failed to generate ICNS.
mv: rename icon.icns to Launch_Vivado.app/Contents/Resources/icon.icns: No such file or directory
```

ichi4096, 2023-05-19:
Are you using the 2023 version of Vivado? Because I haven't tested it myself yet.

ichi4096, 2023-05-23:
I'll fix it this week when I download the new version of Vivado.

PIX3LFLUX, 2023-06-06:
Same issue with M1 Pro, Docker version 24.0.2 (build cb74dfc). Can you please fix it?

benjaminmordaunt, 2023-07-08:
@ichi4096 Any update on this?

ichi4096, 2023-07-27:
Yeah, I‘ll get around it in a few days or so.

hakatu, 2023-08-18:
I got the same problem

```
This is a fresh install.
INFO Could not detect the display scale (hDPI).
       If you are using a high resolution monitor, you can set the insaller scale factor like this: 
       export XINSTALLER_SCALE=2
       setenv XINSTALLER_SCALE 2
Running in batch mode...
Picked up JAVA_TOOL_OPTIONS: -Dsun.java2d.xrender=false
This command is not for general use and should only be run as the result of a call to
ProcessBuilder.start() or Runtime.exec() in a java application
Copyright (c) 1986-2022 Xilinx, Inc.  All rights reserved.
Copyright (c) 2022-2023 Advanced Micro Devices, Inc.  All rights reserved.
INFO  - User has accepted the EULAs. 
ERROR - The value specified in the configuration file for Modules (Virtex UltraScale+ HBM:1,DocNav:1,Kintex UltraScale:1,Artix UltraScale+:1,Spartan-7:1,Artix-7:1,Virtex UltraScale+:1,Vitis Model Composer(Xilinx Toolbox for MATLAB and Simulink. Includes the functionality of System Generator for DSP):1,Zynq UltraScale+ MPSoC:1,Zynq-7000:1,Virtex UltraScale+ 58G:1,Kintex-7:1,Install Devices for Kria SOMs and Starter Kits:1,Kintex UltraScale+:1) is not valid. 

Generating App icon
./install.sh:68: no matches found: Xilinx/Vivado/*/doc/images/vivado_logo.png
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
icon.iconset:Failed to generate ICNS.
mv: rename icon.icns to Launch_Vivado.app/Contents/Resources/icon.icns: No such file or directory
```

I think it's because of the 2023 bin?

dmendat, 2023-08-18:
Looks like I'm trying this out at the same time. I have the exact same issues as @hakatu - I tried 2023 and then 2022.3, both with the issues that are posted above.

dmendat, 2023-08-19:
Today I also tried 2022.1 which has the same problem for me.

dmendat, 2023-08-19:
So I managed to get Vivado to open by editing Launch_Vivado.app/Launch_Vivado. I commented out the `while` line and the `done` lines. Then it doesn't get stuck in a loop and the program opens up. I installed 2022.2 by the way.

However, I do have a Terminal window that pops up having trouble with not seeing a USB device connected:

`ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed`

But I'm not very concerned about that and just hit Control-C in that window. I'll probably program the FPGA using other means. I haven't done more testing, but this may work for me. Thank you and hope this helps!

ichi4096, 2023-08-20:
I've updated the script to support the 2023 version and also added some little bugfixes. Do your linker errors still persist with these updates?

ichi4096, 2023-08-22:
I'm gonna close this as the issue has been fixed with the 2023 version being supported now. The linker errors, if they persist, should be addressed in #15.

closed 2023-08-22 by ichi4096