ichi4096/vivado-on-silicon-mac #9: /Applications/Launch_Vivado.app/Launch_Vivado:12: parse error near `done'

skotskt, 2023-05-31:
When I opened Launch_Vivado.app, it never launched.
I tried to open it with command line, and got 
```
/Applications/Launch_Vivado.app/Launch_Vivado:12: parse error near `done'
```

The content of Launch_Vivado.app is as follows:
```
#!/bin/zsh
open -a XQuartz
open -a Docker
while ! /usr/local/bin/docker ps &> /dev/null
do
open -a Docker
sleep 5
done
while ! [ -f "/tmp/.X11-unix" ]
do\open -a XQuartz
sleep 5
done

/usr/local/bin/docker run --rm --name vivado_container --mount type=bind,source="/tmp/.X11-unix",target="/tmp/.X11-unix" --mount type=bind,source="/Users/kotaro/workspace/dockerfiles/vivado-on-silicon-mac-main",target="/home/user" --platform linux/amd64 x64-linux sudo -H -u user bash /home/user/start_vivado.sh &
osascript -e 'tell app "Terminal" to do script " while ! [[ $(ps aux | grep vivado_container | wc -l | tr -d \"\\n\\t \") == \"1\" ]]; do /Users/kotaro/workspace/dockerfiles/vivado-on-silicon-mac-main/xvcd/bin/xvcd; sleep 1; done; exit"'
```

I removed '\' in line 10 and inserted a line break on the same place and then it seems to be an endless loop between ```while ! [ -f "/tmp/.X11-unix" ]``` and ```done```.

I would appreciate any help you can give me!

justinbasko, 2023-06-11:
In Installation, looking at step 3, did you get the instructions (in yellow) in your terminal before opening Launch_Vivado.app? My terminal is stuck at the first yellow instructions "Launching docker daemon and quartz..." and then I'm running into the same issue you're describing.

dmendat, 2023-08-19:
I'm also having the same issue as shown here when I try to manually run Launch_Vivado.

dmendat, 2023-08-19:
So I managed to get Vivado to open by editing Launch_Vivado.app/Launch_Vivado. I commented out the `while` line and the `done` lines. Then it doesn't get stuck in a loop and the program opens up. I installed 2022.2 by the way.

However, I do have a Terminal window that pops up having trouble with not seeing a USB device connected:

`ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed`

But I'm not very concerned about that and just hit Control-C in that window. I'll probably program the FPGA using other means. I haven't done more testing, but this may work for me. Thank you and hope this helps!

ichi4096, 2023-08-20:
@dmendat The terminal window will display this message if you launch Vivado and afterwards plug in the FPGA. You can still use the FPGA, the terminal just doesn't clear the error messages automatically. Also, keep in mind that the software responsible for communicating with the FPGA has a limited set of supported boards.

@skotskt I'm sorry, I must have made a typo in the script, but it should be fixed now. The '\' was supposed to be a '\n'.

closed 2023-08-20 by ichi4096

dmendat, 2023-08-20:
Great, thank you for this and all your work! I saw your other messages about fixing things and updating to support 2023, so I'll give that a shot soon and report back. :) I'm excited about the sims hopefully working as well.