ichi4096/vivado-on-silicon-mac #11: app never launched

YunaSung, 2023-06-16:
When I opened Launch_Vivado.app, it never launched.

`Generating App icon
./install.sh:68: no matches found: Xilinx/Vivado/*/doc/images/vivado_logo.png
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
icon.iconset:Failed to generate ICNS.
mv: rename icon.icns to Launch_Vivado.app/Contents/Resources/icon.icns: No such file or directory
./install.sh:85: no such file or directory: Launch_Vivado.app/Launch_Vivado
./install.sh:87: no such file or directory: Launch_Vivado.app/Launch_Vivado
./install.sh:89: no such file or directory: Launch_Vivado.app/Launch_Vivado
chmod: Launch_Vivado.app/Launch_Vivado: No such file or directory
`
How can i solve it?

ichi4096, 2023-07-27:
Are you using the newest version of Vivado? I haven‘t come around updating my scripts for it yet.

ichi4096, 2023-08-20:
I have updated the repo such that the 2023 version is supported. If this issue persists, you should provide more of the terminal ouput so I can pin down the cause.

closed 2023-08-20 by ichi4096