ichi4096/vivado-on-silicon-mac #12: Windows randomly disappear - esp. with modal dialogs

benjaminmordaunt, 2023-07-09:
When performing various actions in the GUI, such as starting a synthesis run, or opening the hardware manager, modal dialogs are displayed with a status/progress bar. However, the main GUI window (IDE window) does becomes hidden and does not return.

I can make the window return under the following conditions:
- Do not use the '+' icon to maximise the IDE window.
- Use XQuartz in 'Full Screen mode'
- Unfocus and refocus the full-screen X11 "window" - then the IDE reappears.

This seems really buggy. Is there a fix for this?

ichi4096, 2023-08-20:
I think that's an XQuartz issue and I'll probably switch to using some method where the window manager runs inside the VM and that gets forwarded via VNC or some spice-tool.

ichi4096, 2023-09-23:
I will rewrite the tool such that it uses VNC and the remote desktop tool built into macOS.

closed 2024-03-29 by ichi4096