ichi4096/vivado-on-silicon-mac #14: Segfault from JRE during install

BreadMoncher, 2023-08-05:
I have been getting a segfault when attempting the install. Issue occurred with both 2023.1 and 2022.2 web installers. Perhaps a bug either with the java packages which come with the installer or the x86 emulator? Currently running Sonoma, so could be a problem with my operating system.

I noticed segfaults occur somewhat sporadically (sometimes almost immediately, sometimes a few minutes into the install).

System specs: running on M1 Max, Sonoma 23A5301h, Docker 4.21.1

Terminal output:

```
You successfully logged into your account. The installation will begin now.
If a window pops up, simply close it to finish the installation.
This is a fresh install.
Running in batch mode...
Picked up JAVA_TOOL_OPTIONS: -Dsun.java2d.xrender=false
Copyright (c) 1986-2023 Xilinx, Inc.  All rights reserved.
INFO  - User has accepted the EULAs. 
WARN  - You don't have permissions to create shortcuts for all users.In-order to create shortcuts for all users, you must start the installer in "elevated" mode . 
INFO  - Internet connection validated, can connect to internet. 
INFO  - Authenticated user ajcantu@crimson.ua.edu successfully. 
INFO  - Installing Edition: Vivado ML Standard 
INFO  - Installation directory is /home/user/Xilinx 

Downloading files (7.44 MB / 19.38 GB) 33 h and 49 m(s) left at 200 KB/sec. /#                    
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x00007ffffdd9a462, pid=137, tid=165
#
# JRE version: OpenJDK Runtime Environment 18.9 (11.0.11+9) (build 11.0.11+9)
# Java VM: OpenJDK 64-Bit Server VM 18.9 (11.0.11+9, mixed mode, tiered, compressed oops, g1 gc, linux-amd64)
# Problematic frame:
# V  [libjvm.so+0x79a462]  G1ParScanThreadState::copy_to_survivor_space(InCSetState, oopDesc*, markOopDesc*)+0x42
#
# No core dump will be written. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# /home/user/hs_err_pid137.log
#
# If you would like to submit a bug report, please visit:
#   https://bugreport.java.com/bugreport/crash.jsp
#
/home/user/installer/xsetup: line 98:   137 Aborted                 ${X_JAVA_HOME}/bin/java ${ARGS} -cp ${X_CLASS_PATH} com.xilinx.installer.api.InstallerLauncher "$@"
Generating App icon
./install.sh:68: no matches found: Xilinx/Vivado/*/doc/images/vivado_logo.png
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
icon.iconset:Failed to generate ICNS.
mv: rename icon.icns to Launch_Vivado.app/Contents/Resources/icon.icns: No such file or directory
```

ichi4096, 2023-08-13:
I have just freshly installed the 2022.2 version and had no problem (M2, macOS 13.5). The JRE crashing is very odd, especially during the download phase, and makes me think it is a Rosetta related issue (maybe because you are using a beta version of macOS). Since the error message is memory access related, have you tried increasing the allocated swap or memory in the Docker settings?

ichi4096, 2023-08-20:
I can't reproduce this issue and it is probably related to you using a beta version of macOS, so I'll close this issue.

closed 2023-08-20 by ichi4096

DavidJRobertson, 2025-04-23:
I am running into this problem too. I am running macOS Sonoma 14.5 on M1 Pro, docker v4.40.0.

```
INFO  - Installing Edition: Vivado ML Standard
INFO  - Installation directory is /home/user/Xilinx

Downloading files (948.26 MB / 20.62 GB) 1 h and 28 m(s) left at 4 MB/sec. /#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x00007ffffe7d8166, pid=162, tid=166
#
# JRE version: OpenJDK Runtime Environment Temurin-21.0.1+12 (21.0.1+12) (build 21.0.1+12-LTS)
# Java VM: OpenJDK 64-Bit Server VM Temurin-21.0.1+12 (21.0.1+12-LTS, mixed mode, sharing, tiered, compressed oops, compressed class ptrs, g1 gc, linux-amd64)
# Problematic frame:
Downloading files (948.50 MB / 20.62 GB) 1 h and 28 m(s) left at 4 MB/sec. -  void OopOopIterateDispatch<G1CMOopClosure>::Table::oop_oop_iterate<InstanceKlass, narrowOop>(G1CMOopClosure*, oopDesc*, Klass*)+0x446
#
# No core dump will be written. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
Downloading files (949.07 MB / 20.62 GB) 1 h and 28 m(s) left at 4 MB/sec. \# An error report file with more information is saved as:
# /tmp/hs_err_pid162.log
Downloading files (949.72 MB / 20.62 GB) 1 h and 28 m(s) left at 4 MB/sec. /#
# If you would like to submit a bug report, please visit:
#   https://github.com/adoptium/adoptium-support/issues
#
/home/user/installer/xsetup: line 115:   162 Aborted                 ${X_JAVA_HOME}/bin/java ${ARGS} -cp ${X_CLASS_PATH} com.xilinx.installer.api.InstallerLauncher "$@"
An error occurred during installation. Please run cleanup.sh and try again.
```