ichi4096/vivado-on-silicon-mac #15: Permission issues and Java application issue

dmendat, 2023-08-20:
Here is the output of my Terminal that shows the errors I'm seeing so far. I will see if the rest of the install works regardless though since it didn't stop running.

`$ caffeinate -dim ./install.sh
Launching Docker daemon and XQuartz...
Building Docker image
ERROR: open /Users/user/.docker/buildx/activity/desktop-linux: permission denied
Drag and drop the installation binary into this terminal window and press Enter: 
/Users/user/Downloads/Xilinx_Unified_2023.1_0507_1903_Lin64.bin 
Launching Docker container and installation script
Found Installer
2023
Extracting installer
Creating directory /home/user/installer
Verifying archive integrity... All good.
Uncompressing AMD Installer for FPGAs and Adaptive SoCs................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................../usr/bin/chown: changing ownership of './data/unified_xilinx_eulas_onpremise_vivado.txt': Permission denied
/usr/bin/chown: changing ownership of './data/unified_xilinx_eulas_bootgen.txt': Permission denied
/usr/bin/chown: changing ownership of './data/unified_3rd_party_eulas_labtool.txt': Permission denied
/usr/bin/chown: changing ownership of './data/unified_3rd_party_eulas_vivado.txt': Permission denied
/usr/bin/chown: changing ownership of './data/unified_xilinx_eulas_hwserver.txt': Permission denied
/usr/bin/chown: changing ownership of './data/unified_3rd_party_eulas.txt': Permission denied
/usr/bin/chown: changing ownership of './data/vitisembedded_3rd_party_eulas.txt': Permission denied
/usr/bin/chown: changing ownership of './data/unified_3rd_party_eulas_onpremise_vitis.txt': Permission denied
/usr/bin/chown: changing ownership of './data/unified_xilinx_eulas.txt': Permission denied
/usr/bin/chown: changing ownership of './data/unified_3rd_party_eulas_hwserver.txt': Permission denied
/usr/bin/chown: changing ownership of './data/unified_3rd_party_eulas_bootgen.txt': Permission denied
/usr/bin/chown: changing ownership of './data/unified_xilinx_eulas_labtool.txt': Permission denied
/usr/bin/chown: changing ownership of './data/vitisembedded_eulas.txt': Permission denied
/usr/bin/chown: changing ownership of './data/unified_xilinx_eulas_docnav.txt': Permission denied
/usr/bin/chown: changing ownership of './data/unified_xilinx_eulas_onpremise_vitis.txt': Permission denied
/usr/bin/chown: changing ownership of './data/unified_xilinx_eulas_vitis.txt': Permission denied
/usr/bin/chown: changing ownership of './data/unified_3rd_party_eulas_modelcomposer.txt': Permission denied
/usr/bin/chown: changing ownership of './data/unified_3rd_party_eulas_docnav.txt': Permission denied
/usr/bin/chown: changing ownership of './data/unified_xilinx_eulas_modelcomposer.txt': Permission denied
/usr/bin/chown: changing ownership of './data/unified_3rd_party_eulas_vitis.txt': Permission denied
/usr/bin/chown: changing ownership of './data/unified_xilinx_eulas_vivado.txt': Permission denied
/usr/bin/chown: changing ownership of './data/unified_3rd_party_eulas_onpremise_vivado.txt': Permission denied
/usr/bin/chgrp: changing group of './data/unified_xilinx_eulas_onpremise_vivado.txt': Permission denied
/usr/bin/chgrp: changing group of './data/unified_xilinx_eulas_bootgen.txt': Permission denied
/usr/bin/chgrp: changing group of './data/unified_3rd_party_eulas_labtool.txt': Permission denied
/usr/bin/chgrp: changing group of './data/unified_3rd_party_eulas_vivado.txt': Permission denied
/usr/bin/chgrp: changing group of './data/unified_xilinx_eulas_hwserver.txt': Permission denied
/usr/bin/chgrp: changing group of './data/unified_3rd_party_eulas.txt': Permission denied
/usr/bin/chgrp: changing group of './data/vitisembedded_3rd_party_eulas.txt': Permission denied
/usr/bin/chgrp: changing group of './data/unified_3rd_party_eulas_onpremise_vitis.txt': Permission denied
/usr/bin/chgrp: changing group of './data/unified_xilinx_eulas.txt': Permission denied
/usr/bin/chgrp: changing group of './data/unified_3rd_party_eulas_hwserver.txt': Permission denied
/usr/bin/chgrp: changing group of './data/unified_3rd_party_eulas_bootgen.txt': Permission denied
/usr/bin/chgrp: changing group of './data/unified_xilinx_eulas_labtool.txt': Permission denied
/usr/bin/chgrp: changing group of './data/vitisembedded_eulas.txt': Permission denied
/usr/bin/chgrp: changing group of './data/unified_xilinx_eulas_docnav.txt': Permission denied
/usr/bin/chgrp: changing group of './data/unified_xilinx_eulas_onpremise_vitis.txt': Permission denied
/usr/bin/chgrp: changing group of './data/unified_xilinx_eulas_vitis.txt': Permission denied
/usr/bin/chgrp: changing group of './data/unified_3rd_party_eulas_modelcomposer.txt': Permission denied
/usr/bin/chgrp: changing group of './data/unified_3rd_party_eulas_docnav.txt': Permission denied
/usr/bin/chgrp: changing group of './data/unified_xilinx_eulas_modelcomposer.txt': Permission denied
/usr/bin/chgrp: changing group of './data/unified_3rd_party_eulas_vitis.txt': Permission denied
/usr/bin/chgrp: changing group of './data/unified_xilinx_eulas_vivado.txt': Permission denied
/usr/bin/chgrp: changing group of './data/unified_3rd_party_eulas_onpremise_vivado.txt': Permission denied

Log into your Xilinx account to download the necessary files.
This is a fresh install.
INFO Could not detect the display scale (hDPI).
       If you are using a high resolution monitor, you can set the insaller scale factor like this: 
       export XINSTALLER_SCALE=2
       setenv XINSTALLER_SCALE 2
Running in batch mode...
Picked up JAVA_TOOL_OPTIONS: -Dsun.java2d.xrender=false
This command is not for general use and should only be run as the result of a call to
ProcessBuilder.start() or Runtime.exec() in a java application
Copyright (c) 1986-2022 Xilinx, Inc.  All rights reserved.
[... 36 lines cut ...]
INFO  - Installation directory is /home/user/Xilinx 

This command is not for general use and should only be run as the result of a call to
ProcessBuilder.start() or Runtime.exec() in a java application
WARN  - Unable to get OS info, will use Install platform. 
java.util.concurrent.ExecutionException: Error occurred executing the command: [/bin/sh, -c, cat /etc/os-release |grep PRETTY_NAME]
	at com.xilinx.installer.utils.b.a(Unknown Source)
	at com.xilinx.installer.utils.b.a(Unknown Source)
	at com.xilinx.installer.utils.s.E(Unknown Source)
	at j.t.a(Unknown Source)
	at n.j.a(Unknown Source)
	at n.j.a(Unknown Source)
	at n.k.run(Unknown Source)
Caused by: java.io.IOException: Cannot run program "/bin/sh": error=0, Failed to exec spawn helper: pid: 180, exit value: 1
	at java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1143)
	at java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1073)
	... 7 common frames omitted
Caused by: java.io.IOException: error=0, Failed to exec spawn helper: pid: 180, exit value: 1
	at java.base/java.lang.ProcessImpl.forkAndExec(Native Method)
	at java.base/java.lang.ProcessImpl.<init>(ProcessImpl.java:314)
	at java.base/java.lang.ProcessImpl.start(ProcessImpl.java:244)
	at java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1110)
	... 8 common frames omitted`

Thank you so much!

ichi4096, 2023-08-20:
You're welcome :-)

The permissions errors seem odd. Have you followed the instructions exactly and removed the old vivado folder and extracted it from the newly downloaded zip? If so, they somehow don't appear on my machine...

As for the Java errors, I saw them myself but the installation was successful, so they're low on my priority list.

Anyway, wish me luck for my Digital Design exam tomorrow (that subject is the reason I even made this repo).

dmendat, 2023-08-21:
Good luck - hope your exam goes well!

Yes, I followed the instructions and started from a fresh download of the repository.

I'm seeing a ton of errors still (had to truncate this output a bit so it would be short enough to post):

```
$ caffeinate -dim ./install.sh
Launching Docker daemon and XQuartz...
Building Docker image
ERROR: open /Users/user/.docker/buildx/activity/desktop-linux: permission denied
Drag and drop the installation binary into this terminal window and press Enter: 
/Users/user/Downloads/Xilinx_Unified_2023.1_0507_1903_Lin64.bin 
Launching Docker container and installation script
Found Installer
2023
Extracting installer
Creating directory /home/user/installer
Verifying archive integrity... All good.
Uncompressing AMD Installer for FPGAs and Adaptive SoCs................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................../usr/bin/chown: changing ownership of './data/unified_xilinx_eulas_onpremise_vivado.txt': Permission denied
/usr/bin/chown: changing ownership of './data/unified_xilinx_eulas_bootgen.txt': Permission denied
/usr/bin/chown: changing ownership of './data/unified_3rd_party_eulas_labtool.txt': Permission denied
/usr/bin/chown: changing ownership of './data/unified_3rd_party_eulas_vivado.txt': Permission denied
/usr/bin/chown: changing ownership of './data/unified_xilinx_eulas_hwserver.txt': Permission denied
/usr/bin/chown: changing ownership of './data/unified_3rd_party_eulas.txt': Permission denied
/usr/bin/chown: changing ownership of './data/vitisembedded_3rd_party_eulas.txt': Permission denied
/usr/bin/chown: changing ownership of './data/unified_3rd_party_eulas_onpremise_vitis.txt': Permission denied
/usr/bin/chown: changing ownership of './data/unified_xilinx_eulas.txt': Permission denied
/usr/bin/chown: changing ownership of './data/unified_3rd_party_eulas_hwserver.txt': Permission denied
/usr/bin/chown: changing ownership of './data/unified_3rd_party_eulas_bootgen.txt': Permission denied
/usr/bin/chown: changing ownership of './data/unified_xilinx_eulas_labtool.txt': Permission denied
/usr/bin/chown: changing ownership of './data/vitisembedded_eulas.txt': Permission denied
/usr/bin/chown: changing ownership of './data/unified_xilinx_eulas_docnav.txt': Permission denied
/usr/bin/chown: changing ownership of './data/unified_xilinx_eulas_onpremise_vitis.txt': Permission denied
/usr/bin/chown: changing ownership of './data/unified_xilinx_eulas_vitis.txt': Permission denied
/usr/bin/chown: changing ownership of './data/unified_3rd_party_eulas_modelcomposer.txt': Permission denied
/usr/bin/chown: changing ownership of './data/unified_3rd_party_eulas_docnav.txt': Permission denied
/usr/bin/chown: changing ownership of './data/unified_xilinx_eulas_modelcomposer.txt': Permission denied
/usr/bin/chown: changing ownership of './data/unified_3rd_party_eulas_vitis.txt': Permission denied
/usr/bin/chown: changing ownership of './data/unified_xilinx_eulas_vivado.txt': Permission denied
/usr/bin/chown: changing ownership of './data/unified_3rd_party_eulas_onpremise_vivado.txt': Permission denied
/usr/bin/chgrp: changing group of './data/unified_xilinx_eulas_onpremise_vivado.txt': Permission denied
/usr/bin/chgrp: changing group of './data/unified_xilinx_eulas_bootgen.txt': Permission denied
/usr/bin/chgrp: changing group of './data/unified_3rd_party_eulas_labtool.txt': Permission denied
/usr/bin/chgrp: changing group of './data/unified_3rd_party_eulas_vivado.txt': Permission denied
/usr/bin/chgrp: changing group of './data/unified_xilinx_eulas_hwserver.txt': Permission denied
/usr/bin/chgrp: changing group of './data/unified_3rd_party_eulas.txt': Permission denied
/usr/bin/chgrp: changing group of './data/vitisembedded_3rd_party_eulas.txt': Permission denied
/usr/bin/chgrp: changing group of './data/unified_3rd_party_eulas_onpremise_vitis.txt': Permission denied
/usr/bin/chgrp: changing group of './data/unified_xilinx_eulas.txt': Permission denied
/usr/bin/chgrp: changing group of './data/unified_3rd_party_eulas_hwserver.txt': Permission denied
/usr/bin/chgrp: changing group of './data/unified_3rd_party_eulas_bootgen.txt': Permission denied
/usr/bin/chgrp: changing group of './data/unified_xilinx_eulas_labtool.txt': Permission denied
/usr/bin/chgrp: changing group of './data/vitisembedded_eulas.txt': Permission denied
/usr/bin/chgrp: changing group of './data/unified_xilinx_eulas_docnav.txt': Permission denied
/usr/bin/chgrp: changing group of './data/unified_xilinx_eulas_onpremise_vitis.txt': Permission denied
/usr/bin/chgrp: changing group of './data/unified_xilinx_eulas_vitis.txt': Permission denied
/usr/bin/chgrp: changing group of './data/unified_3rd_party_eulas_modelcomposer.txt': Permission denied
[... cut ...]
/Users/user/Downloads/vivado-on-silicon-mac-main/Xilinx/Vivado/2023.1/doc/images/vivado_logo.png
  /Users/user/Downloads/vivado-on-silicon-mac-main/icon.iconset/icon_256x256.png
/Users/user/Downloads/vivado-on-silicon-mac-main/Xilinx/Vivado/2023.1/doc/images/vivado_logo.png
  /Users/user/Downloads/vivado-on-silicon-mac-main/icon.iconset/icon_256x256@2x.png
/Users/user/Downloads/vivado-on-silicon-mac-main/Xilinx/Vivado/2023.1/doc/images/vivado_logo.png
  /Users/user/Downloads/vivado-on-silicon-mac-main/icon.iconset/icon_512x512.png
```

dmendat, 2023-08-21:
And then if I try to run the application, I get an error saying that "this application is not supported on this Mac."

ichi4096, 2023-08-21:
Are you an administrator and have full read/write privileges to the Vivado folder?

Could you provide the file contents of Launch_Vivado.app/Launch_Vivado ?

dmendat, 2023-08-21:
Yes, I am an admin and have full read/write privileges to the main Vivado folder (I assume you mean vivado-on-silicon-mac-main?).

In vivado-on-silicon-mac-main/installer/data the permissions look like this:

```
-r-xr-xr-x   1 user  staff    4905268 May 10 16:44 unified_3rd_party_eulas.txt
-r-xr-xr-x   1 user  staff    1076418 May 10 16:44 unified_3rd_party_eulas_bootgen.txt
-r-xr-xr-x   1 user  staff     593975 May 10 16:44 unified_3rd_party_eulas_docnav.txt
-r-xr-xr-x   1 user  staff     873280 May 10 16:44 unified_3rd_party_eulas_hwserver.txt
-r-xr-xr-x   1 user  staff    2044843 May 10 16:44 unified_3rd_party_eulas_labtool.txt
-r-xr-xr-x   1 user  staff     623446 May 10 16:44 unified_3rd_party_eulas_modelcomposer.txt
-r-xr-xr-x   1 user  staff    5594516 May 10 16:44 unified_3rd_party_eulas_onpremise_vitis.txt
-r-xr-xr-x   1 user  staff    3781807 May 10 16:44 unified_3rd_party_eulas_onpremise_vivado.txt
-rwxr-xr-x@  1 user  staff     267353 May  7 17:23 unified_3rd_party_eulas_p4.txt
-r-xr-xr-x   1 user  staff    5594516 May 10 16:44 unified_3rd_party_eulas_vitis.txt
-r-xr-xr-x   1 user  staff    3781807 May 10 16:44 unified_3rd_party_eulas_vivado.txt
```

Here is Launch_Vivado.app/Launch_Vivado:

```
#!/bin/zsh
open -a XQuartz
open -a Docker
while ! /usr/local/bin/docker ps &> /dev/null
do
open -a Docker
sleep 5
done
while ! [ -d "/tmp/.X11-unix" ]
do
open -a XQuartz
sleep 5
done

/usr/local/bin/docker run --rm --name vivado_container --mount type=bind,source="/tmp/.X11-unix",target="/tmp/.X11-unix" --mount type=bind,source="/Users/user/Downloads/vivado-on-silicon-mac-main",target="/home/user" --platform linux/amd64 x64-linux sudo -H -u user bash /home/user/start_vivado.sh &
osascript -e 'tell app "Terminal" to do script " while ! [[ $(ps aux | grep vivado_container | wc -l | tr -d \"\\n\\t \") == \"1\" ]]; do /Users/user/Downloads/vivado-on-silicon-mac-main/xvcd/bin/xvcd; sleep 1; done; exit"'
```

ichi4096, 2023-08-21:
Something about your "Downloads" folder seems odd: The `..` in your output has an `@` as its last permission when it should actually be `+` according to [this site](https://apple.stackexchange.com/questions/26776/what-is-the-mark-at-the-end-of-file-description). Also, it seems that Docker is struggling to get the [necessary permissions](https://stackoverflow.com/questions/75686903/open-users-user-docker-buildx-current-permission-denied-on-macbook) in your user folder (see the first few lines in your output). Are the permissions of your user folder in order?

My `ls` output (notice the `-e` flag):
```
michi@macBook vivado-on-silicon-mac-main % ls -lae
total 544736
drwxrwxr-x@  15 michi  staff        480 21 Aug 22:23 .
drwx------+ 280 michi  staff       8960 21 Aug 22:21 ..
 0: group:everyone deny delete
-rw-rw-r--@   1 michi  staff         10 20 Aug 20:13 .gitignore
-rw-rw-r--@   1 michi  staff       1902 20 Aug 20:13 Dockerfile
-rw-rw-r--@   1 michi  staff       7048 20 Aug 20:13 LICENSE
drwxrwxr-x@   4 michi  staff        128 20 Aug 20:13 Launch_Vivado.app
-rw-rw-r--@   1 michi  staff       5982 20 Aug 20:13 README.md
-rwxr-xr-x@   1 michi  staff  278858784 21 Aug 22:22 Xilinx_Unified_2023.1_0507_1903_Lin64.bin
-rwxr-xr-x@   1 michi  staff       1753 20 Aug 20:13 docker.sh
-rwxr-xr-x@   1 michi  staff       3655 20 Aug 20:13 install.sh
-rw-rw-r--@   1 michi  staff       1611 20 Aug 20:13 install_config_2022.txt
-rw-rw-r--@   1 michi  staff       1623 20 Aug 20:13 install_config_2023.txt
drwxr-xr-x@   8 michi  staff        256 21 Aug 22:22 installer
-rwxr-xr-x@   1 michi  staff        523 20 Aug 20:13 start_vivado.sh
drwxrwxr-x@   7 michi  staff        224 20 Aug 20:13 xvcd
```

My output when installing:
```
michi@macBook vivado-on-silicon-mac-main % caffeinate -dim ./install.sh
Launching Docker daemon and XQuartz...
Building Docker image
[+] Building 1.3s (14/14) FINISHED                         docker:desktop-linux
 => [internal] load build definition from Dockerfile                       0.0s
 => => transferring dockerfile: 2.02kB                                     0.0s
 => [internal] load .dockerignore                                          0.0s
 => => transferring context: 2B                                            0.0s
 => [internal] load metadata for docker.io/library/ubuntu:latest           1.2s
 => [ 1/10] FROM docker.io/library/ubuntu@sha256:ec050c32e4a6085b423d36ec  0.0s
 => CACHED [ 2/10] RUN apt update && apt upgrade -y                        0.0s
 => CACHED [ 3/10] RUN apt install -y --no-install-recommends --allow-una  0.0s
 => CACHED [ 4/10] RUN apt install -y --no-install-recommends --allow-una  0.0s
 => CACHED [ 5/10] RUN useradd --create-home --shell /bin/bash --user-gro  0.0s
 => CACHED [ 6/10] RUN sh -c 'echo "user:pass" | chpasswd'                 0.0s
 => CACHED [ 7/10] RUN chown -R user:user /home/user                       0.0s
 => CACHED [ 8/10] RUN mkdir -p /home/user/.config/pcmanfm/LXDE/           0.0s
 => CACHED [ 9/10] RUN ln -sf /usr/local/share/doro-lxde-wallpapers/deskt  0.0s
 => CACHED [10/10] RUN sed -i '/en_US.UTF-8/s/^# //g' /etc/locale.gen &&   0.0s
 => exporting to image                                                     0.0s
 => => exporting layers                                                    0.0s
 => => writing image sha256:671088afe694d988b4830ea65e2d6b0dd0a77fa2cae5b  0.0s
 => => naming to docker.io/library/x64-linux                               0.0s

What's Next?
  View summary of image vulnerabilities and recommendations → docker scout quickview
Drag and drop the installation binary into this terminal window and press Enter: 
/Users/michi/Downloads/Xilinx_Unified_2023.1_0507_1903_Lin64.bin 
Launching Docker container and installation script
Found Installer
2023
Extracting installer
Creating directory /home/user/installer
Verifying archive integrity... All good.
Uncompressing AMD Installer for FPGAs and Adaptive SoCs..................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................
Log into your Xilinx account to download the necessary files.
This is a fresh install.
INFO Could not detect the display scale (hDPI).
       If you are using a high resolution monitor, you can set the insaller scale factor like this: 
       export XINSTALLER_SCALE=2
       setenv XINSTALLER_SCALE 2
Running in batch mode...
Picked up JAVA_TOOL_OPTIONS: -Dsun.java2d.xrender=false
This command is not for general use and should only be run as the result of a call to
ProcessBuilder.start() or Runtime.exec() in a java application
Copyright (c) 1986-2022 Xilinx, Inc.  All rights reserved.
Copyright (c) 2022-2023 Advanced Micro Devices, Inc.  All rights reserved.
[... cut ...]
ProcessBuilder.start() or Runtime.exec() in a java application
ERROR - There was an error executing: [/bin/sh, -c, echo $TMPDIR] 
INFO  - Internet connection validated, can connect to internet. 
INFO  - In order to generate the authentication token please provide your AMD account E-mail Address and password.
 
E-mail Address:
```

Maybe try the installation in a different folder, updating macOS and Docker to the newest versions. Also, try deleting the older containers / images in the Docker dashboard.

ichi4096, 2023-08-24:
@dmendat maybe try `chown -R ~/Downloads/vivado-on-silicon-mac-main/.` and `chown -R ~/.docker/.` and then the installation procedure

dmendat, 2023-08-24:
Thank you for your help, and apologies about the slow reply.

I ran the chown commands, but it wouldn't run from the ~/Downloads folder. Then I ran things in a new folder (~/Documents). MacOS is fully updated. I also tried reinstalling Docker after deleting as many traces of it as I could from my system, but it does look like there is an update so I am installing that now.

I'm still having some issues (seen below), but Vivado does run again now after installing in the ~/Documents folder.

```
$ caffeinate -dim ./install.sh
Launching Docker daemon and XQuartz...
Building Docker image
[+] Building 0.3s (14/14) FINISHED                                                                     docker:desktop-linux
 => [internal] load .dockerignore                                                                                      0.0s
 => => transferring context: 2B                                                                                        0.0s
 => [internal] load build definition from Dockerfile                                                                   0.0s
 => => transferring dockerfile: 2.02kB                                                                                 0.0s
 => [internal] load metadata for docker.io/library/ubuntu:latest                                                       0.3s
 => [ 1/10] FROM docker.io/library/ubuntu@sha256:ec050c32e4a6085b423d36ecd025c0d3ff00c38ab93a3d71a460ff1c44fa6d77      0.0s
 => CACHED [ 2/10] RUN apt update && apt upgrade -y                                                                    0.0s
 => CACHED [ 3/10] RUN apt install -y --no-install-recommends --allow-unauthenticated dbus dbus-x11 x11-utils xorg al  0.0s
 => CACHED [ 4/10] RUN apt install -y --no-install-recommends --allow-unauthenticated python3-pip python3-dev build-e  0.0s
 => CACHED [ 5/10] RUN useradd --create-home --shell /bin/bash --user-group --groups adm,sudo user                     0.0s
 => CACHED [ 6/10] RUN sh -c 'echo "user:pass" | chpasswd'                                                             0.0s
 => CACHED [ 7/10] RUN chown -R user:user /home/user                                                                   0.0s
 => CACHED [ 8/10] RUN mkdir -p /home/user/.config/pcmanfm/LXDE/                                                       0.0s
 => CACHED [ 9/10] RUN ln -sf /usr/local/share/doro-lxde-wallpapers/desktop-items-0.conf /home/user/.config/pcmanfm/L  0.0s
 => CACHED [10/10] RUN sed -i '/en_US.UTF-8/s/^# //g' /etc/locale.gen &&     locale-gen                                0.0s
 => exporting to image                                                                                                 0.0s
 => => exporting layers                                                                                                0.0s
 => => writing image sha256:3765dbfcb1b564be5eb2ca1a9e8520f26bfa6d7b9545b715811321aefa1f45bc                           0.0s
 => => naming to docker.io/library/x64-linux                                                                           0.0s

What's Next?
  View summary of image vulnerabilities and recommendations → docker scout quickview
Drag and drop the installation binary into this terminal window and press Enter: 
/Users/user/Downloads/Xilinx_Unified_2023.1_0507_1903_Lin64.bin 
Launching Docker container and installation script
Installer was previously extracted. Removing the extracted directory.
Found Installer
2023
Extracting installer
Creating directory /home/user/installer
Verifying archive integrity... All good.
Uncompressing AMD Installer for FPGAs and Adaptive SoCs................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................../usr/bin/chown: changing ownership of './data/unified_xilinx_eulas_onpremise_vivado.txt': Permission denied
/usr/bin/chown: changing ownership of './data/unified_xilinx_eulas_bootgen.txt': Permission denied
/usr/bin/chown: changing ownership of './data/unified_3rd_party_eulas_labtool.txt': Permission denied
/usr/bin/chown: changing ownership of './data/unified_3rd_party_eulas_vivado.txt': Permission denied
/usr/bin/chown: changing ownership of './data/unified_xilinx_eulas_hwserver.txt': Permission denied
/usr/bin/chown: changing ownership of './data/unified_3rd_party_eulas.txt': Permission denied
/usr/bin/chown: changing ownership of './data/vitisembedded_3rd_party_eulas.txt': Permission denied
/usr/bin/chown: changing ownership of './data/unified_3rd_party_eulas_onpremise_vitis.txt': Permission denied
/usr/bin/chown: changing ownership of './data/unified_xilinx_eulas.txt': Permission denied
/usr/bin/chown: changing ownership of './data/unified_3rd_party_eulas_hwserver.txt': Permission denied
[... cut ...]
/Users/user/Documents/vivado-on-silicon-mac-main/Xilinx/Vivado/2023.1/doc/images/vivado_logo.png
  /Users/user/Documents/vivado-on-silicon-mac-main/icon.iconset/icon_256x256@2x.png
/Users/user/Documents/vivado-on-silicon-mac-main/Xilinx/Vivado/2023.1/doc/images/vivado_logo.png
  /Users/user/Documents/vivado-on-silicon-mac-main/icon.iconset/icon_512x512.png
```

I'm going to try this again after removing one image that is in my Docker dashboard and updating Docker. I'll let you know later on how things go.

ichi4096, 2023-08-24:
No worries. What would be interesting are the permissions within the container. Could you run `/usr/local/bin/docker run -it --rm --mount type=bind,source="$script_dir",target="/home/user" --platform linux/amd64 x64-linux bash -c "ls -lah /home/user/data` with $script_dir replaced with the absolute path of the Vivado folder on your Mac? Maybe this could give some insight into the permissions issue.

dmendat, 2023-08-24:
Sure. I will run that later today after the install finishes again. I will be out for a bit though while it is running, so I'll get back to you tonight. Thanks!

dmendat, 2023-08-25:
So you mean run it like this?

```
vivado-on-silicon-mac-main user$ /usr/local/bin/docker run -it --rm --mount type=bind,source="/Users/user/Documents/vivado-on-silicon-mac-main",target="/home/user" --platform linux/amd64 x64-linux bash -c "ls -lah /home/user/data"
/usr/bin/ls: cannot access '/home/user/data': No such file or directory
```

ichi4096, 2023-08-25:
Exactly

dmendat, 2023-08-26:
Ah yes, it looks like this may be the issue:

```
$ /usr/local/bin/docker run -it --rm --mount type=bind,source="/Users/user/Documents/vivado-on-silicon-mac-main",target="/home/user" --platform linux/amd64 x64-linux bash -c "ls -lah /home/user/Xilinx"
total 12K
drwxr-xr-x 10 root root  320 Aug 24 21:36 .
drwxrwxr-x 22 root root  704 Aug 26 02:30 ..
drwxr-xr-x 16 root root  512 Aug 24 21:54 DocNav
drwxr-xr-x  2 root root   64 Aug 24 21:54 Downloads
-rw-r--r--  1 root root 8.1K Aug 24 23:25 .DS_Store
drwxr-xr-x  4 root root  128 Aug 24 21:39 Model_Composer
drwxr-xr-x  4 root root  128 Aug 24 21:38 Vitis_HLS
drwxr-xr-x  4 root root  128 Aug 24 21:36 Vivado
drwxr-xr-x  9 root root  288 Aug 24 21:50 xic
drwxr-xr-x  6 root root  192 Aug 24 21:37 .xinstall
```

I am new to Docker, so I'm not sure how to resolve this even though it's pretty simple.

ichi4096, 2023-09-23:
Sorry for not writing in a while. I will rewrite the script such that it gets the directory permissions as an installation requirement.

dmendat, 2023-11-06:
Well, I'm not sure whether things have changed or what, but I upgraded my OS recently and redownloaded this repo. I have now been able to run Vivado with no issues so far, and I've been even programming my board from Vivado.

Thank you so much for your code and help!

closed 2023-11-06 by dmendat