ichi4096/vivado-on-silicon-mac #20: Open Launch_Vivado.app with non-default folder

gri11, 2023-09-04:
My goal now is connect Vivado with Basys 3 Board via USB.
I'm installed Vivado completed in folder name `vivado-on-silicon-mac-main 3`
I ran it. Got this error.
[image]
So, I use Launch_Vivado.app in broken installation folder name `vivado-on-silicon-mac-main`
And got this error.
[image]

Is there anyway that I can solve these problem?

ichi4096, 2023-09-04:
Maybe there is an error because the path of the directory contains a space? Anyway, permission denied errors for the xvcd binary could possibly stem from you not allowing its execution in the Settings app, or because the file permissions aren't set correctly. Try `chmod +x xvcd` in a terminal.

The "device not found" errors are to be expected whenever the board isn't plugged in. When you do plug in the board, there is no success message and the errors aren't cleared. If the errors keep repeating, try `lsusb` in a terminal and see if your board is even picked up by your Mac.

dmendat, 2023-09-04:
It's possible that the dongle isn't working well with the Mac? I've definitely had issues in the past with some USB dongles/hubs not working with embedded devices, for example.

ichi4096, 2023-09-04:
I could get it to work with a dongle, the problem was with the micro usb cables, since a lot of them only have the power wires and no data wires.

dmendat, 2023-09-04:
Yes that's definitely another common issue.

gri11, 2023-09-04:
Oh I see, so I could try to connect my board with micro usb cable with power wires and data wires right?

ichi4096, 2023-09-04:
Yes. xvcd could only connect to your board if it shows up in lsusb.

gri11, 2023-09-05:
There're still error that existed when I already found the board in local machine:
[image]

The error:
[image]

ichi4096, 2023-09-05:
Try first plugging in the board, then check with `lsusb` to make sure it's connected properly (the "Future Technology Devices International" device) and then opening up Vivado. If that doesn't work, does the xvcd terminal window display error messages?

gri11, 2023-09-05:
For now this and error after I use [openFPGALoader](https://github.com/trabucayre/openFPGALoader) to load bitstring to the board
`bind: Address already in use`
[image]

ichi4096, 2023-09-05:
What command did you use exactly? Also make sure that no instance of xvcd is running since it will make the board unavalailable to any other program.

gri11, 2023-09-05:
I use this command
`openFPGALoader -b basys3 main.bit` # Loading in SRAM
and
`openFPGALoader -b basys3 -f main.bit` # Writing in flash

gri11, 2023-09-05:
No, when exited Vivado, xvcd will be exited too.
[image]

james-kollross, 2023-09-13:
I'm having a very similar issue with getting the hardware manager to connect to my Basys 3. It shows up in lsusb and in System Information, albeit only if I connect it through a USB hub and not directly with a dongle.
 
[image]

lsusb result:
[image]

ichi4096, 2023-09-23:
I'm rewriting the script such that the docker instance will be logged in an open terminal window while xvcd has a little window of it's own, indicating that it is running / connected.

ichi4096, 2024-07-28:
That rewrite has arrived about a year later : /
Please try the updated scripts. If the problems persist, open another issue.

closed 2024-07-28 by ichi4096