ichi4096/vivado-on-silicon-mac #21: Throw some light here

gaurav3015, 2023-09-04:
after launching vivado.app i get these lines of feedback on terminal. Can anyone guide me through this. Thank you

while ! [[ $(ps aux | grep vivado_container | wc -l | tr -d "\n\t ") == "1" ]]; do /Users/gauravbadgujar/Downloads/vivado-on-silicon-mac-main/xvcd/bin/xvcd; sleep 1; done; exit
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed
ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed

Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.

[Process completed]

ichi4096, 2023-09-04:
These error messages occur as long as your fpga board isn't plugged in. If you plug it in, the errors don't get automatically cleared. Does the actual Vivado program start?

gaurav3015, 2023-09-04:
Program does not start. I did tried 2-3 times.
The vivado app doesn't have a icon but only white square.
In spotlight search there are 2 vivado apps showing. 1 as silicon-mac-main and 2 as applications.
after opening any one of them Xquards and terminal shows up in the docker and sometimes vivado too for few seconds. but eventually the application doen't opens

gaurav3015, 2023-09-06:
Now when i launch the vivado-on-silicon-mac-main/Launch_Vivado.app 3 terminal windows are opening. 1 from Xterm and 2 from the terminal app itself.

while the 3rd terminal shows this mac@user ~ %  while ! [[ $(ps aux | grep vivado_container | wc -l | tr -d "\n\t ") == "1" ]]; do /Users/<path>/vivado-on-silicon-mac-main/xvcd/bin/xvcd; sleep 1; done; exit

Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.
Deleting expired sessions...none found.

[Process completed]

What is happening?
Please help @ichi4096

QQAI13, 2023-09-16:
Hi, I got the same errors here after installation. (M1 pro / 2022.1 version vivado bin file)
I want to ensure that the vivado program will only start when I plug in the FPGA board?
When I do not plug it in, it will only show the error like this?
[image]
Thank you!

gaurav3015, 2023-09-17:
@QQAI13 i dont have any board around so couldn't test. But I'm still stuck at the same error. Have you found any solution?

Thanks

ichi4096, 2023-09-23:
There are multiple errors mentioned in this issue. `ftdi_usb_open(0x0403, 0x6010): -3 (device not found)
io_init failed` is an error message that is expected to show up whenever the fpga isn't connected. It does not directly mean that Vivado couldn't connect with the fpga. If other errors persist (such as the app not launching at all), then please consider adding to an existing issue or create a new issue.

closed 2023-09-23 by ichi4096