ichi4096/vivado-on-silicon-mac #23: xvcd can't be opened due to Apple not being able to check for malicious software

bhuiyah, 2023-10-04:
Hi, every time I try to open vivado, it works fine but I will get this pop-up.
[image]
when I press show in finder, i get this: 
[image]
Is there anyway I can fix this issue?
[image]

ichi4096, 2023-10-06:
Stay tuned, I'm making a signed version of xvcd.

ichi4096, 2024-03-29:
Ok, so I'm not able to sign it. There should theoretically be no problem running xvcd when done using terminal scripts, which is what I'm currently working on. Sorry for the delay.

closed 2024-03-29 by ichi4096