ichi4096/vivado-on-silicon-mac #24: App doesn't launch at all

kbc0, 2023-10-09:
I did the installing according to README, but when I try to open Launch_Vivado, XQuartz and Terminal launches, but after that, well... nothing happens. This may be a dumb issue, but I couldn't make the Vivado open.

Here is the terminal output: 
[image]

Thanks in advance :)

laserroger, 2023-10-13:
I bet your installation is not successful, which usually something like 

```
Downloading files (165.03 MB / 21.86 GB) 20 minutes left at 22 MB/sec. -#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x00007ffffe221290, pid=151, tid=183
#
# JRE version: OpenJDK Runtime Environment Temurin-17.0.3+7 (17.0.3+7) (build 17.0.3+7)
# Java VM: OpenJDK 64-Bit Server VM Temurin-17.0.3+7 (17.0.3+7, mixed mode, sharing, tiered, compressed oops, compressed class ptrs, g1 gc, linux-amd64)
# Problematic frame:
# V  [libjvm.so+0x733290]  G1ParScanThreadState::trim_queue_to_threshold(unsigned int)+0x35b0
#
# No core dump will be written. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# /home/user/hs_err_pid151.log
#
# If you would like to submit a bug report, please visit:
#   https://github.com/adoptium/adoptium-support/issues
#
/home/user/installer/xsetup: line 115:   151 Aborted                 ${X_JAVA_HOME}/bin/java ${ARGS} -cp ${X_CLASS_PATH} com.xilinx.installer.api.InstallerLauncher "$@"
Generating App icon
./install.sh:68: no matches found: Xilinx/Vivado/*/doc/images/vivado_logo.png
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
icon.iconset:Failed to generate ICNS.
mv: rename icon.icns to Launch_Vivado.app/Contents/Resources/icon.icns: No such file or directory
``` 

happens. I'm also trying to figure this out.

laserroger, 2023-10-13:
BTW, I had some luck by installing 2022 instead of 2023 version. Maybe you want to check that as well?

viniciuslagrota, 2023-10-14:
I had the same issue here for both 2022 and 2023. MacOS Sonoma and M2.

I bypassed the error setting the Docker to use only one CPU. 

2023 successfully installed.

VladBara, 2023-12-02:
same issue. I think it has something to do with the "permissions denied" in the beginning of the this log file:

"Verifying archive integrity... All good.
Uncompressing AMD Installer for FPGAs and Adaptive SoCs......................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................chown: changing ownership of './data/unified_xilinx_eulas_onpremise_vivado.txt': Permission denied
chown: changing ownership of './data/unified_xilinx_eulas_bootgen.txt': Permission denied
chown: changing ownership of './data/unified_xilinx_eulas_p4.txt': Permission denied
chown: changing ownership of './data/unified_3rd_party_eulas_labtool.txt': Permission denied
chown: changing ownership of './data/unified_3rd_party_eulas_vivado.txt': Permission denied
chown: changing ownership of './data/unified_xilinx_eulas_hwserver.txt': Permission denied
chown: changing ownership of './data/unified_3rd_party_eulas.txt': Permission denied
chown: changing ownership of './data/vitisembedded_3rd_party_eulas.txt': Permission denied
chown: changing ownership of './data/unified_3rd_party_eulas_onpremise_vitis.txt': Permission denied
chown: changing ownership of './data/unified_xilinx_eulas.txt': Permission denied
chown: changing ownership of './data/unified_3rd_party_eulas_hwserver.txt': Permission denied
chown: changing ownership of './data/unified_3rd_party_eulas_bootgen.txt': Permission denied
chown: changing ownership of './data/unified_xilinx_eulas_labtool.txt': Permission denied
chown: changing ownership of './data/vitisembedded_eulas.txt': Permission denied
chown: changing ownership of './data/unified_xilinx_eulas_docnav.txt': Permission denied
chown: changing ownership of './data/unified_xilinx_eulas_onpremise_vitis.txt': Permission denied
chown: changing ownership of './data/unified_xilinx_eulas_vitis.txt': Permission denied
chown: changing ownership of './data/unified_3rd_party_eulas_modelcomposer.txt': Permission denied
chown: changing ownership of './data/unified_3rd_party_eulas_docnav.txt': Permission denied
chown: changing ownership of './data/pdm_3rd_party_eulas.txt': Permission denied
chown: changing ownership of './data/unified_xilinx_eulas_modelcomposer.txt': Permission denied
chown: changing ownership of './data/unified_3rd_party_eulas_p4.txt': Permission denied
chown: changing ownership of './data/pdm_xilinx_eulas.txt': Permission denied
chown: changing ownership of './data/unified_3rd_party_eulas_vitis.txt': Permission denied
chown: changing ownership of './data/unified_xilinx_eulas_vivado.txt': Permission denied
chown: changing ownership of './data/unified_3rd_party_eulas_onpremise_vivado.txt': Permission denied
chgrp: changing group of './data/unified_xilinx_eulas_onpremise_vivado.txt': Permission denied
chgrp: changing group of './data/unified_xilinx_eulas_bootgen.txt': Permission denied
chgrp: changing group of './data/unified_xilinx_eulas_p4.txt': Permission denied
chgrp: changing group of './data/unified_3rd_party_eulas_labtool.txt': Permission denied
chgrp: changing group of './data/unified_3rd_party_eulas_vivado.txt': Permission denied
chgrp: changing group of './data/unified_xilinx_eulas_hwserver.txt': Permission denied
chgrp: changing group of './data/unified_3rd_party_eulas.txt': Permission denied
chgrp: changing group of './data/vitisembedded_3rd_party_eulas.txt': Permission denied
chgrp: changing group of './data/unified_3rd_party_eulas_onpremise_vitis.txt': Permission denied
chgrp: changing group of './data/unified_xilinx_eulas.txt': Permission denied
chgrp: changing group of './data/unified_3rd_party_eulas_hwserver.txt': Permission denied
chgrp: changing group of './data/unified_3rd_party_eulas_bootgen.txt': Permission denied
chgrp: changing group of './data/unified_xilinx_eulas_labtool.txt': Permission denied
chgrp: changing group of './data/vitisembedded_eulas.txt': Permission denied
chgrp: changing group of './data/unified_xilinx_eulas_docnav.txt': Permission denied
chgrp: changing group of './data/unified_xilinx_eulas_onpremise_vitis.txt': Permission denied
chgrp: changing group of './data/unified_xilinx_eulas_vitis.txt': Permission denied
chgrp: changing group of './data/unified_3rd_party_eulas_modelcomposer.txt': Permission denied
chgrp: changing group of './data/unified_3rd_party_eulas_docnav.txt': Permission denied
chgrp: changing group of './data/pdm_3rd_party_eulas.txt': Permission denied
chgrp: changing group of './data/unified_xilinx_eulas_modelcomposer.txt': Permission denied
chgrp: changing group of './data/unified_3rd_party_eulas_p4.txt': Permission denied
chgrp: changing group of './data/pdm_xilinx_eulas.txt': Permission denied
chgrp: changing group of './data/unified_3rd_party_eulas_vitis.txt': Permission denied
[... cut ...]
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
icon.iconset:Failed to generate ICNS.
mv: rename icon.icns to Launch_Vivado.app/Contents/Resources/icon.icns: No such file or directory
test@tests-MacBook-Pro vivado-on-silicon-mac-main %

Cipppp, 2023-12-20:
Anyone solved this problem i also get this logs

Launching Docker daemon and XQuartz...
Building Docker image
[+] Building 1.2s (15/15) FINISHED                         docker:desktop-linux
 => [internal] load build definition from Dockerfile                       0.0s
 => => transferring dockerfile: 2.02kB                                     0.0s
 => [internal] load .dockerignore                                          0.0s
 => => transferring context: 2B                                            0.0s
 => [internal] load metadata for docker.io/library/ubuntu:latest           1.2s
 => [auth] library/ubuntu:pull token for registry-1.docker.io              0.0s
 => [ 1/10] FROM docker.io/library/ubuntu@sha256:6042500cf4b44023ea1894ef  0.0s
 => CACHED [ 2/10] RUN apt update && apt upgrade -y                        0.0s
 => CACHED [ 3/10] RUN apt install -y --no-install-recommends --allow-una  0.0s
 => CACHED [ 4/10] RUN apt install -y --no-install-recommends --allow-una  0.0s
 => CACHED [ 5/10] RUN useradd --create-home --shell /bin/bash --user-gro  0.0s
 => CACHED [ 6/10] RUN sh -c 'echo "user:pass" | chpasswd'                 0.0s
 => CACHED [ 7/10] RUN chown -R user:user /home/user                       0.0s
 => CACHED [ 8/10] RUN mkdir -p /home/user/.config/pcmanfm/LXDE/           0.0s
 => CACHED [ 9/10] RUN ln -sf /usr/local/share/doro-lxde-wallpapers/deskt  0.0s
 => CACHED [10/10] RUN sed -i '/en_US.UTF-8/s/^# //g' /etc/locale.gen &&   0.0s
 => exporting to image                                                     0.0s
 => => exporting layers                                                    0.0s
 => => writing image sha256:f78140b1c61fa2dea1bf98f4802c3539bcafabcac4eba  0.0s
 => => naming to docker.io/library/x64-linux                               0.0s

What's Next?
  View a summary of image vulnerabilities and recommendations → docker scout quickview
Drag and drop the installation binary into this terminal window and press Enter:
/Users/ciprianpirvu/Downloads/Xilinx\*.bin
Launching Docker container and installation script
Found Installer
2022
Extracting installer
Creating directory /home/user/installer
Verifying archive integrity... All good.
Uncompressing Xilinx Installer..................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................
Log into your Xilinx account to download the necessary files.
This is a fresh install.
Running in batch mode...
Picked up JAVA_TOOL_OPTIONS: -Dsun.java2d.xrender=false
Copyright (c) 1986-2023 Xilinx, Inc.  All rights reserved.
INFO  - Internet connection validated, can connect to internet.
INFO  - In order to generate the authentication token please provide your Xilinx account E-mail Address and password.

E-mail Address:pirvuciprian37@gmail.com
Password:
INFO  - Generating authentication token...
INFO  - Saved authentication token file successfully, valid until 12/27/2023 01:55 PM

You successfully logged into your account. The installation will begin now.
If a window pops up, simply close it to finish the installation.
This is a fresh install.
Running in batch mode...
Picked up JAVA_TOOL_OPTIONS: -Dsun.java2d.xrender=false
Copyright (c) 1986-2023 Xilinx, Inc.  All rights reserved.
INFO  - User has accepted the EULAs.
ERROR - There is not enough disk space on /home/user/Xilinx to install.

Generating App icon
./install.sh:68: no matches found: Xilinx/Vivado/*/doc/images/vivado_logo.png
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
icon.iconset:Failed to generate ICNS.
mv: rename icon.icns to Launch_Vivado.app/Contents/Resources/icon.icns: No such file or directory

VladBara, 2023-12-29:
> Anyone solved this problem? I also get this logs :(
> 
> Launching Docker daemon and XQuartz... Building Docker image [+] Building 1.2s (15/15) FINISHED docker:desktop-linux => [internal] load build definition from Dockerfile 0.0s => => transferring dockerfile: 2.02kB 0.0s => [internal] load .dockerignore 0.0s => => transferring context: 2B 0.0s => [internal] load metadata for docker.io/library/ubuntu:latest 1.2s => [auth] library/ubuntu:pull token for registry-1.docker.io 0.0s => [ 1/10] FROM docker.io/library/ubuntu@sha256:6042500cf4b44023ea1894ef 0.0s => CACHED [ 2/10] RUN apt update && apt upgrade -y 0.0s => CACHED [ 3/10] RUN apt install -y --no-install-recommends --allow-una 0.0s => CACHED [ 4/10] RUN apt install -y --no-install-recommends --allow-una 0.0s => CACHED [ 5/10] RUN useradd --create-home --shell /bin/bash --user-gro 0.0s => CACHED [ 6/10] RUN sh -c 'echo "user:pass" | chpasswd' 0.0s => CACHED [ 7/10] RUN chown -R user:user /home/user 0.0s => CACHED [ 8/10] RUN mkdir -p /home/user/.config/pcmanfm/LXDE/ 0.0s => CACHED [ 9/10] RUN ln -sf /usr/local/share/doro-lxde-wallpapers/deskt 0.0s => CACHED [10/10] RUN sed -i '/en_US.UTF-8/s/^# //g' /etc/locale.gen && 0.0s => exporting to image 0.0s => => exporting layers 0.0s => => writing image sha256:f78140b1c61fa2dea1bf98f4802c3539bcafabcac4eba 0.0s => => naming to docker.io/library/x64-linux 0.0s
> 
> What's Next? View a summary of image vulnerabilities and recommendations → docker scout quickview Drag and drop the installation binary into this terminal window and press Enter: /Users/ciprianpirvu/Downloads/Xilinx*.bin Launching Docker container and installation script Found Installer 2022 Extracting installer Creating directory /home/user/installer Verifying archive integrity... All good. Uncompressing Xilinx Installer.................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................. Log into your Xilinx account to download the necessary files. This is a fresh install. Running in batch mode... Picked up JAVA_TOOL_OPTIONS: -Dsun.java2d.xrender=false Copyright (c) 1986-2023 Xilinx, Inc. All rights reserved. INFO - Internet connection validated, can connect to internet. INFO - In order to generate the authentication token please provide your Xilinx account E-mail Address and password.
> 
> E-mail Address:[pirvuciprian37@gmail.com](mailto:pirvuciprian37@gmail.com) Password: INFO - Generating authentication token... INFO - Saved authentication token file successfully, valid until 12/27/2023 01:55 PM
> 
> You successfully logged into your account. The installation will begin now. If a window pops up, simply close it to finish the installation. This is a fresh install. Running in batch mode... Picked up JAVA_TOOL_OPTIONS: -Dsun.java2d.xrender=false Copyright (c) 1986-2023 Xilinx, Inc. All rights reserved. INFO - User has accepted the EULAs. ERROR - There is not enough disk space on /home/user/Xilinx to install.
> 
> Generating App icon ./install.sh:68: no matches found: Xilinx/Vivado/*/doc/images/vivado_logo.png Warning: not a valid file - skipping Warning: not a valid file - skipping Warning: not a valid file - skipping Warning: not a valid file - skipping Warning: not a valid file - skipping Warning: not a valid file - skipping Warning: not a valid file - skipping Warning: not a valid file - skipping Warning: not a valid file - skipping icon.iconset:Failed to generate ICNS. mv: rename icon.icns to Launch_Vivado.app/Contents/Resources/icon.icns: No such file or directory

Hello, [Cipppp](https://github.com/Cipppp) , I managed to do it by following this fix: 

https://github.com/ichi4096/vivado-on-silicon-mac/issues/29

Also, be careful with the name of the folder:

https://github.com/ichi4096/vivado-on-silicon-mac/issues/27

https://github.com/ichi4096/vivado-on-silicon-mac/issues/30

In the end, it worked for me. It runs. I have an M1 , 16 GB .

vjedith, 2024-05-04:
> I had the same issue here for both 2022 and 2023. MacOS Sonoma and M2.
> 
> I bypassed the error setting the Docker to use only one CPU.
> 
> 2023 successfully installed.

 ...... by the way can wo show ho did you do it ( because I have same model)

ichi4096, 2024-07-28:
There is very little I can do about segfaults during installation because I cannot reproduce them.  The other problems should be resolved now since I've updated the installation scripts to correctly set permissions and support newer versions of Vivado.

closed 2024-07-28 by ichi4096