ichi4096/vivado-on-silicon-mac #29: Bind mount does not provide the required permissions for the installer

davidchisnall, 2023-11-09:
The installer needs to be able to chown files in the working directory.  This is not possible with a bind mount from the host.  Copying files into the container in the first step of docker.sh avoids this problem.

riccardilorenzo, 2023-11-11:
Hi! I'm facing this same issue. Can you better explain how to solve this issue please?

davidchisnall, 2023-11-11:
I will try to write up what I did, but unfortunately I worked with a download (as recommended in the README) not a git clone, so it's going to take a while to figure out what I changed.

I ended up running the installer in graphical mode, but that also required fixing things because bind mounting the UNIX domain socket didn't work and I needed to use TCP.

For the Arty A7, the shipped version of xvcd didn't work either, but I have managed to patch that to make it work.

DaryaGolovko, 2023-11-21:
Hi, I have the same problem. Can you please write how you fixed it?

NelsonDane, 2023-11-29:
@DaryaGolovko If you take a look at [my fork](https://github.com/NelsonDane/vivado-on-silicon-mac), I solved this issue by extracting and taking ownership of the files in an internal docker directory before moving them to the mountpoint

NelsonDane, 2023-12-21:
Awesome, I created a PR with my changes

VladBara, 2023-12-28:
OMG, it works. Be blessed !!! 
This fix should be added to the main branch and also mentioned in the description.

mohay3069, 2024-01-17:
> @DaryaGolovko If you take a look at [my fork](https://github.com/NelsonDane/vivado-on-silicon-mac), I solved this issue by extracting and taking ownership of the files in an internal docker directory before moving them to the mountpoint

doesnt worK:

errors:
1)ERROR - There was an error executing: [/bin/sh, -c, echo $TMPDIR] 

2)ProcessBuilder.start() or Runtime.exec() in a java application
Copyright (c) 1986-2022 Xilinx, Inc.  All rights reserved.
Copyright (c) 2022-2024 Advanced Micro Devices, Inc.  All rights reserved.
INFO  - User has accepted the EULAs. 
ERROR - There is not enough disk space on /home/user/Xilinx to install. 

Generating App icon
./install.sh:73: no matches found: Xilinx/Vivado/*/doc/images/vivado_logo.png
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
icon.iconset:Failed to generate ICNS.
mv: rename icon.icns to Launch_Vivado.app/Contents/Resources/icon.icns: No such file or directory

NelsonDane, 2024-01-17:
It says you don't have enough disk space

mohay3069, 2024-01-17:
> It says you don't have enough disk space @mohay3069
INFO  - User has accepted the EULAs. 
ERROR - There is not enough disk space on /home/user/Xilinx to install. 

Generating App icon
./install.sh:73: no matches found: Xilinx/Vivado/*/doc/images/vivado_logo.png
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
Warning:  not a valid file - skipping
icon.iconset:Failed to generate ICNS.
mv: rename icon.icns to Launch_Vivado.app/Contents/Resources/icon.icns: No such file or directory
i have 100 gibabytes of space how much more do i need?

mohay3069, 2024-01-17:
> It says you don't have enough disk space @mohay3069

is this it?

is this it:{
  "builder": {
    "gc": {
      "defaultKeepStorage": "20GB",
      "enabled": true
    }
  },
  "experimental": false
}

ichi4096, 2024-03-29:
Isn't all data not within a bind mount lost anyway? How could you do an install to a directory in the container that is not a  bind mount and have it be permanent?

davidchisnall, 2024-03-29:
No, a bind mount is a host directory that is mounted into the container. The scripts install into that directory. It is not ephemeral.

ichi4096, 2024-07-28:
I've updated the scripts, addressing both the Vivado version and permissions issues. Please try them and open new issues if there are any further problems.

closed 2024-07-28 by ichi4096