ichi4096/vivado-on-silicon-mac #37: [Tip] Launching Vitis

reteps, 2024-03-04:
Hi there, I was able to launch Vitis by swapping libgdk to version 3. It appears there is a known incompatibility if libgdk version 2 and JavaFX are loaded simultaneously.

So my full script is:

```bash
#!/bin/bash
cd /home/user
export LD_PRELOAD="/lib/x86_64-linux-gnu/libudev.so.1 /lib/x86_64-linux-gnu/libselinux.so.1 /lib/x86_64-linux-gnu/libz.so.1 /lib/x86_64-linux-gnu/libgdk-3.so.0"
export JAVA_TOOL_OPTIONS="-Dsun.java2d.xrender=false"
export JAVA_OPTS="-Dsun.java2d.xrender=false"
export DISPLAY="host.docker.internal:0"
/home/user/Xilinx/Vitis_HLS/*/settings64.sh
/home/user/Xilinx/Vitis_HLS/*/bin/vitis_hls
```

Note that I am using `/lib/x86_64-linux-gnu/libgdk-3.so.0` instead of `/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0`

[image]

ichi4096, 2024-03-29:
Nice! I have added libgtk2 because without preloading that Vivado crashes.

sapertuz, 2024-04-15:
Are you also able to run the Vitis Unified IDE ?

reteps, 2024-04-15:
I have not tried, but I assume no as it is not included in the installation package by default.

ichi4096, 2024-07-29:
> Are you also able to run the Vitis Unified IDE ?

You can try to install it like You would on Linux otherwise (see the updated README). Make sure to install into the /home/user/Xilinx directory.

ichi4096, 2024-08-29:
I haven't considered running software other than Vivado yet, however, if You want to run other programs and see their errors, You'll have better luck launching them from the terminal.

sapertuz, 2024-08-30:
I do launch them from the terminal... it creates a .jou and .log files which are helpful for this. sadly Vitis doesnt create anything :(

sapertuz, 2024-10-14:
Btw, In case anyone is interested:

Installing `libnss3 libasound2 libsecret-1-0` allows Docker Container to open Vitis Unified IDE

Tested on MacOS Sequoia 15.0.1 on M3-pro MacBook Pro and Docker Desktop 4.34.3 (170107)

parya03, 2024-11-08:
> Btw, In case anyone is interested:
> 
> Installing `libnss3 libasound2 libsecret-1-0` allows Docker Container to open Vitis Unified IDE
> 
> @ichi4096 maybe you can include the packages on the Dockerfile ?
> 
> Tested on MacOS Sequoia 15.0.1 on M3-pro MacBook Pro and Docker Desktop 4.34.3 (170107)
> 
> EDIT: After rebuilding the Docker image adding these lines:
> 
> ```dockerfile
> # install dependencies for Vitis
> RUN apt-get install -y --no-install-recommends \
>     libnss3 libasound2 libsecret-1-0
> ```
> 
> I could successfully synthesize and export an IP in Vitis Unified IDE :)

Thanks so much for this! I was able to launch Vitis Unified IDE.

However, after a little bit (running a build, etc) Vitis hangs and does not recover. It seems to happen more if I change the window (resize, etc). Not sure if anyone also has this issue.

mistune, 2025-02-22:
@sapertuz Thanks this works

sapertuz, 2025-02-25:
To anyone with the same problem, as https://github.com/ichi4096/vivado-on-silicon-mac/issues/37#issuecomment-2465621307 

My solution so far has been working with Vitis using the interactive python shell `vitis -I`.
Building from the python shell doesn't halt vitis

seanstone, 2025-02-28:
I have the same problem as https://github.com/ichi4096/vivado-on-silicon-mac/issues/37#issuecomment-2411580909

This is not a solution, however I found that the underlying Electron dev tools still work when Vitis freezes up, but the HTML Elements shows as empty, and the dev console does not respond:

[image]