ichi4096/vivado-on-silicon-mac #39: Your Account info seems to be wrong!!!

AmanShukla111, 2024-04-21:
Launching Docker daemon and XQuartz...
Building Docker image
[+] Building 799.2s (15/15) FINISHED                       docker:desktop-linux
 => [internal] load build definition from Dockerfile                       0.0s
 => => transferring dockerfile: 1.99kB                                     0.0s
 => [internal] load metadata for docker.io/library/ubuntu:latest           3.3s
 => [auth] library/ubuntu:pull token for registry-1.docker.io              0.0s
 => [internal] load .dockerignore                                          0.0s
 => => transferring context: 2B                                            0.0s
 => CACHED [ 1/10] FROM docker.io/library/ubuntu:latest@sha256:1b8d8ff477  0.0s
 => [ 2/10] RUN apt update && apt upgrade -y                              96.4s
 => [ 3/10] RUN apt install -y --no-install-recommends --allow-unauthen  557.6s
 => [ 4/10] RUN apt install -y --no-install-recommends --allow-unauthen  138.4s 
 => [ 5/10] RUN useradd --create-home --shell /bin/bash --user-group --gr  0.1s 
 => [ 6/10] RUN sh -c 'echo "user:pass" | chpasswd'                        0.2s 
 => [ 7/10] RUN chown -R user:user /home/user                              0.1s 
 => [ 8/10] RUN mkdir -p /home/user/.config/pcmanfm/LXDE/                  0.1s 
 => [ 9/10] RUN ln -sf /usr/local/share/doro-lxde-wallpapers/desktop-item  0.1s 
 => [10/10] RUN sed -i '/en_US.UTF-8/s/^# //g' /etc/locale.gen &&     loc  1.1s 
 => exporting to image                                                     1.6s
 => => exporting layers                                                    1.6s
 => => writing image sha256:b2bddb7c171f77773a411933281038e191764420389d2  0.0s
 => => naming to docker.io/library/x64-linux                               0.0s

What's Next?
  View a summary of image vulnerabilities and recommendations → docker scout quickview
Drag and drop the installation binary into this terminal window and press Enter: 
/Users/amanshukla/Downloads/FPGAs_AdaptiveSoCs_Unified_2023.2_1013_2256_Lin64.bin 
Launching Docker container and installation script
Found Installer
md5sum: '/home/user/Xilinx*.bin': No such file or directory
2022
Extracting installer
chmod: cannot access '/home/user/Xilinx*.bin': No such file or directory
/home/user/docker.sh: line 53: /home/user/Xilinx*.bin: No such file or directory
Log into your Xilinx account to download the necessary files.
/home/user/docker.sh: line 57: /home/user/installer/xsetup: No such file or directory
Your account information seems to be wrong. Please try logging in again.
/home/user/docker.sh: line 57: /home/user/installer/xsetup: No such file or directory
Your account information seems to be wrong. Please try logging in again.
/home/user/docker.sh: line 57: /home/user/installer/xsetup: No such file or directory
Your account information seems to be wrong. Please try logging in again.
/home/user/docker.sh: line 57: /home/user/installer/xsetup: No such file or directory

What's this about and how do I fix this?

hahacharlie, 2024-04-22:
Experiencing same issue here.

hahacharlie, 2024-04-22:
I think I figured out. The issue is because we are trying to installing Vivado 2023.2, which I believe is not currently supported yet by the project. Try install 2023.1 instead.

yangch1022, 2024-05-18:
That's because the vivado 2023.2 web installer has the name FPGA*.bin. You have to change the Xilinx*.bin to FPGA*.bin.

ichi4096, 2024-07-28:
The explicit binary name You provide is stored now, so this is no longer an issue.

closed 2024-07-28 by ichi4096