ichi4096/vivado-on-silicon-mac #41: Vivado not launching

KrissPS, 2024-04-30:
I had to change x64-linux to ubuntu since that image is no longer available. Then I was able to download it. And now when I try to launch it I get this terminal, the vivado icon pops up for a moment and then disappears. If I am looking in docker it seems to start something for a split second.
[image]
Fast help would be appreciated. I need this for Uni and other students in my module do so aswell.
[image]

Adarsh063, 2024-07-12:
I am getting this type of message and after that I didn't able to launch vivado

Adarsh063, 2024-07-12:
[image]

ichi4096, 2024-07-28:
I'm sorry the update is late, but there was an issue with Rosetta emulation, which made it difficult to effectively rewrite the scripts. This issue should be resolved now since newer versions of Vivado are supported.

closed 2024-07-28 by ichi4096