ichi4096/vivado-on-silicon-mac #45: can this be done using orbstack?

thuvasooriya, 2024-05-16:
docker desktop is not a very efficient program and since this repo is specific to mac i would like to ask something specific to orbstack

i've been using orbstack for a while now and it's one of the best things happened to docker. so it would be great to have instructions accounting orbstack too.

the readme file makes it sound like this can only be run if you have docker desktop installed and i've seen a condition in install.sh that checks for the app. **so is there anything i have to be aware of** when doing this with orbstack instead of docker desktop cause i'm just new to creating all these x11 forwarded machines and emulations. please enlighten me with the potential pitfalls.

i just don't want to download 20 gigs again and i really don't want to install docker desktop on my mac.

ichi4096, 2024-07-29:
I used Docker because the only other free alternative I knew that supported Rosetta emulation was UTM, and it was prone to I/O disk failures at that time when using Rosetta. I have just updated my scripts, moving away from X11 forwarding to VNC, since XQuartz produces a pixelated eyesore. If You want to use OrbStack, You can go through my scripts and remove Docker Desktop specific references. The `docker` command itself shouldn't need to be replaced because it should be exposed by OrbStack. Apart from that, I can't think of anything to be on the lookout for. The script comments and readme should make clear how this tool works, the VNC forwarding is especially commented in detail.

closed 2024-07-29 by ichi4096

thuvasooriya, 2024-07-29:
yup i opened the issue when i had little to no idea about how those scripts work to get some help. thanks for the reply 👍🏽 
i've made some adjustments (shoddy work at best) related to orbstack and installing xquartz through nix in my fork here. https://github.com/thuvasooriya/vivado4mac
i'll look into the vnc thing too. 
thankyou for working on this in public, much love.