ichi4096/vivado-on-silicon-mac #56: Invalid installer hash

FlavioMili, 2024-09-23:
I get this error message after I select the resolution (standard one) whenever I try to install.

ichi4096, 2024-09-23:
Could You share the exact terminal output / screenshot?

The hash of the installation binary You downloaded is checked in the beginning of the setup, so it should be ok. What can happen is when downloading additional content (right after choosing the resolution), that content can be corrupted and [sometimes the installer can't recover from that](https://github.com/ichi4096/vivado-on-silicon-mac/discussions/53#discussioncomment-10583897). In that case, simply retry the installation but make sure to run "cleanup.sh" beforehand or download my repository into a new folder.

FlavioMili, 2024-09-23:
[image]
this is the problem, I also tried to move the file inside the source directory and outside

ichi4096, 2024-09-23:
This seems weird... But I can tell from the terminal output that the cleanup of leftover files hasn't fully taken place. Please delete this folder and download the newest zip file from my repository.

FlavioMili, 2024-09-23:
[image]

now it is like this, previously it was a problem with the name of the file from amd website containing white space

ichi4096, 2024-09-28:
Well... unfortunately You have too little disk space as mentioned in the second last line.

closed 2024-09-28 by ichi4096

FlavioMili, 2024-09-28:
Does it mean that it might need more than 100GB?

ichi4096, 2024-09-28:
It needs around 100GB, possibly more.