ichi4096/vivado-on-silicon-mac #69: failed installs

simon1tan, 2024-12-03:
Twice now, I've gotten to a point where Vivado starts installing and twice, it failed. The first saw some check checksum errors and "Canceling". The second one, I see a "Killed" for some unknown reason after 11.59GB/20.62GB has downloaded.  It seems both times it failed during download which I can't imagine since most downloaders can retry indefinitely.

It would be nice if the installation can continue where it left off as well as some more verbose output as to what failed. Thanks.

victordiges, 2024-12-12:
I'm running into the same problem. I am installing on an external drive formatted to APFS. I'm trying to install 2024.1, I may try an older version as well.
Has anyone found a workaround?

stellatsimi, 2024-12-21:
I get this error ERROR: failed to solve: process "/bin/sh -c apt-get install -y --no-install-recommends     dbus dbus-x11 x11-utils xorg alsa-utils mesa-utils net-tools     libgl1-mesa-dri gtk2-engines lxappearance fonts-droid-fallback sudo firefox     ubuntu-gnome-default-settings curl gnupg lxde arc-theme     gtk2-engines-murrine gtk2-engines-pixbuf gnome-themes-standard nano xterm" did not complete successfully: exit code: 100
Docker image generation failed!

jeremyPlura, 2025-01-21:
I have been running into installation issues too. Previously I had problems getting the web install to work even on a normal x86_64 machine, and resorted to using the giant complete installer package. It would be great if that was an option. The AMD servers do not seem very reliable, and AFAIK their installer does not do partial or continued downloads.

The full-size installer is like 100GB, but you can download it using a browser that will restart and continue downloads, etc.